nefarius/ScpToolkit #177: Installer stuck

Keefaetana, 2016-01-23:
Yesterday I noticed there's a new version of ScpToolkit and decided to try and install it over the old version I already had installed. I downloaded the installer and ran it, but it got stuck at "Preparing...". I then tried running it as Administrator, but it still gets stuck at the same point. Then I tried uninstalling the old version and rebooting, then running the installer again. It's now been stuck at that point for at least half an hour with no sign of ever getting past the preparing phase. Anything I could do to try and get the installer to work?

I'm running Windows 10 with the most recent updates. The last version before the end of the year installed and worked fine, but neither of the versions released in 2016 seem to install now.

nihaomeeow, 2016-01-28:
Same problem here. Just did a clean install of Win7 Pro. The toolkit installed some of the other things first like the Xbox driver but then it got stuck at the message saying it could take up to 5 minutes.

nefarius, 2016-02-06:
This get's fixed in an upcoming version.

closed 2016-04-10 by nefarius

boskio, 2017-05-10:
..was this ever fixed?.. still sat looking at the waiting screen :)

mloo3, 2017-05-26:
I am still getting stuck as well