nefarius/ScpToolkit #179: Several (input) issues

KinanEldari, 2016-01-24:
When using a USB cable, the controls on the D-pad and sticks sometimes register brief up and up-left input before resetting and doing it again. 

When connected through a bluetooth dongle, these issues go away, but are replaced with others. Now the control sticks sometimes stick the input in games, seconds after you have let go. This happens regardless of asynchronus HID reporting toggled on or off (in Assassin's Creed Syndicate when in tight spaces and trying to move or deal with the camera and sometimes Rocket League). 

Now there's also a fun one where I played Helldivers, and pressed the L-stick in any direction and then clicking L3 made the game register up-input from the D-pad or sometimes the Square button.

And the best part? None of this shows up in the game controller properties. There, the whole emulation process works perfectly. ^^;

warmbaguettes, 2016-01-24:
Same happened to me, just use an older version (https://github.com/nefarius/ScpToolkit/issues/165)

KinanEldari, 2016-01-27:
Oh probably. But that kind of defeats the point of software progressing. Hence my report here so Nefarius knows and can perhaps fix it.

nefarius, 2016-02-06:
I'll look Ingo this ASAP.

KinanEldari, 2016-02-12:
The main thing that bothers me about this bug is that I can't tell you what causes it to help you reproduce, and subsequently fix it. Sorry : (

KinanEldari, 2016-04-16:
I've re-tried your drivers every once in a while after updates to see if things changed, Nefarius. Unfortunately, they haven't. So I decided today that I wanted to see if Scarlet.Crush' old drivers worked, and they do (without my dongle, because it isn't supported by those drivers). So there's something in those and in ScpServer_v1.5.5724.38523 that works, but fails in later versions. Maybe that'll help you fix this bug. If you have time to work on the project at all that is. Regardless, thanks.

nefarius, 2016-04-16:
Can you provide more detail? It annoys me to no end when problems appear between commits I can't explain. I'll browse through the changes since that tag once more but I need more info.

What controller exactly do you use? Do you use multiple? Problems with all of them? Current Windows version?

nefarius, 2016-04-16:
One more: does [v1.5.5730.24251](../releases/tag/v1.5.5730.24251) still work for you or already introduce problems?

KinanEldari, 2016-04-16:
I have a genuine PS3 controller that apparently contains a bluetooth chip that your program flags as a fake. Beyond that, the issues immediately pop up. Is there a way for me to log what's happening under the hood? Not just the regular and glitchy input, I mean. Like what's being executed when the glitches occur. 

Also, I would've tried that build, if I could find a working link, but github is surprisingly hard for me to navigate. All I can find is your most recent version and beta drivers. No "previous versions" tab or link anywhere. That or I'm just blind.

nefarius, 2016-04-16:
Uuuh the link I provided takes you right to the download? [Have a direct link to the archive.](https://github.com/nefarius/ScpToolkit/releases/download/v1.5.5730.24251/ScpServer_v1.5.5730.24251.7z)

Getting recognized as fake is an important fact, please execute the `ScpToolkit Debug Info Collector` utility and submit the archive it generates. Should contain all the necessary info to start investigating for me. If it's a genuine controller but recognized as fake the program will behave differently which may cause unexpected behavior!

KinanEldari, 2016-04-16:
Ah yeah, sorry. Forgot to mention the link didn't work for some reason and gave me nonsense. I've downloaded v1.5.5787.40106 to give that a spin and I'll work my way back, see which version stops giving me issues first.

nefarius, 2016-04-16:
What's your controllers MAC address?

KinanEldari, 2016-04-16:
08:A9:5A:8E:B3:0F

KinanEldari, 2016-04-16:
I've run some trials with two versions of yours of the Scp Toolkit, v1.5.5787.40106 and v1.5.5724.38523. Both did fine with Rocket League, causing no issues. But once I tried Helldivers, an issue popped up where using the the left control stick button to run, the input would also recognize Up on the D-pad, plus issues with the accuracy of the right stick. This persisted when I installed Scarlet.Crush' old drivers. So I'm beginning to suspect it's not something you've done. Just something that got enhanced with something you did going from v1.5 to v1.6

Mind you, Rocket League for example still works swimmingly. Helldivers is just apparently good at showing the issues. 

Also, did you have another way for me to log what's happening under the hood?

nefarius, 2016-04-18:
Your controller uses a Bluetooth chip from [AzureWave Technologies Inc.](https://www.adminsub.net/mac-address-finder/AzureWave), did a bit of research and it looks like Sony also uses their chips in some VAIO notebook models so I guess we can assume the controller is genuine. Meanwhile you could try to add the line `GenuineMacAddress = 08:A9:5A` in `ScpControl.ini` below the section `[Host Controller Interface]` (on a "problematic" version ofc.) and see if behavior changes for you.

Again: problems occur only during usage over USB (wired), Bluetooth (wireless) or both?

You can review all the logs in the `Logs` sub-directory yourself using [YALV!](https://yalv.codeplex.com/), if you need more insight it'd probably be better to clone, compile and debug. Since I don't know where to look I can't just throw in more logging everywhere. You can't log logic errors.

KinanEldari, 2016-04-18:
The issues I experienced I described in the very first post. For the sake of testing, I decided to try with the latest 1.7etc beta drivers and adding 08:A9:5A to the list of genuine controllers, and the wired issues were resolved. The remaining issues now are the ones I first initially only experienced with wireless bluetooth connection (and which I also experienced with Scarlet.Crush's version of the wrapper); sticky camera controls in Assassins Creed Syndicate (though less so), and the left stick clicking down also registering D-pad input somehow (generally left or up) and sometimes Square/X in Helldivers.

Now of course this is stating the obvious, but somewhere, the input coming from the PS3 controller is somehow not being translated right. Since I am no programmer (game designer, actually), I have no idea what code I would even need to dive into to help. And I'm sorry about that. : (

juliuzzz, 2017-07-01:
Hope someone reads this. I've been having this problem for some years but now it has become unbearable.
The solution seems simples but I can't find this scpcontrol.ini file anywhere, Can someone please help me?

PizzasAreForMe, 2017-07-04:
@juliuzzz Same. only i have been researching this a while and i dont seem to draw a good solution. anyone here that can give me a simple solution? or explained simple at least?

figaw, 2017-11-27:
I added `GenuineMacAddress = 08:A9:5A` to the `ScpControl.ini` file, and suddenly it stopped doing the "move to top-left every 20 seconds"-jitter. Afterwards I re-installed/over the PS3 controller driver, to make sure the service restarted.

Great work on the driver ❤️ 

The file was located in `C:\Program Files\Nefarius Software Solutions\ScpToolkit` (default installation)

mikegonzalez2k, 2018-01-10:
This solution definitely fixed the issue with my DS3 controller randonly pulling left as well.
Nefarius, could you please add that Genuine MAC address to the ini in future versions of SCPToolkit so others won't have this problem?

closed 2018-01-26 by nefarius