nefarius/ScpToolkit #180: Sony DualShock 4 Found but not connecting

n3krO, 2016-01-24:
So, I installed ScpToolkit and after that I installed the Bluetooth driver using ScpDriverInstaller to replace the original driver.

I had my controller connected to the dongle when I did that and it disconnected and I can no longer pair the controller to the dongle using Windows 10 pairing screen.

Now when I press the playstation button in my controller it starts flashing and then the ScpToolkit Information tab opens saying "Sony DualShock 4 Found" and the controller turns off.

n3krO, 2016-01-24:
Here's the log from ScpServer:

[image]

WhiteZeroX, 2016-01-24:
I have basically the same issue. I can pair my three DS3 controllers just fine, see them in SCP Monitor normally and use them as XInput controllers. I can plug my DS4 into USB, it shows up and works as XInput as well. But when I try to use the DS4 over Bluetooth, I get the "Sony DualShock 4 found" but it dosen't work at all otherwise. Here's my logs and similar screenshot to OP's. 

[ScpToolkit_Log-Package_Nick_1453654623.zip](https://github.com/nefarius/ScpToolkit/files/102437/ScpToolkit_Log-Package_Nick_1453654623.zip)

n3krO, 2016-01-28:
It seems the bluetooth dongle was broken, already sent to replace.

closed 2016-01-28 by n3krO