nefarius/ScpToolkit #181: Every certain amount of time my DualShock 3 moves the left stick on its own.

r4tzt4r, 2016-01-24:
I'm running Windows 10 (x64). So far I tried the latest release and some of the oldest and I still have this issue. Everything seems normal but suddenly the controller´s left analog stick starts like rapidly moving itself. it registers movement even when I'm not touching it and most buttons stop working. This lasts almost a minute and then it works again. Five minutes later the issue comes back.

Any idea what's causing this? So far ScpToolkit is the only way I've been able to make DS3 work on my W10.  Thanks in advance!

ElephantGent, 2016-01-24:
Have the same problem with both thumbsticks on Win7. Never used to happen.

mirhl, 2016-01-25:
Duplicate of #165

r4tzt4r, 2016-01-25:
So, to solve this I had to go back to ScpServer v1.5.5724.38523. First I uninstalled any other drivers I had, Including those of the  DS3controller. So when I plugged in it was like plugging for the first time. Then I downloaded the older Scarlet.Crush 1.2.0.160 Input Wrapper, then I put the files in the folder "C:/Program Files (x86)/Scarlet.Crush Productions", which it has to be created. Without installing anything, I added to that same folder the files of v1.5.5724.38523, replacing and adding some other files. 

Now, before running SCPDriver.exe, you have to run SCPserver.exe. After opening SCPServer you can install the drivers like you normally would with SCPDriver.exe. This worked for me. And remember that anytime you want to use the controller you have to open (and keep in the background) SCPServer.

closed 2018-01-26 by nefarius