nefarius/ScpToolkit #184: Virtual Bus Driver preinstallation failure.

GmrLeon, 2016-01-26:
I spent much of this morning trying to sort this out on my own with no luck whatsoever. After finding that Scarlet Crush's old drivers had failed to maintain compatibility with Windows 10, I decided to try this tool since it's more actively maintained, but it still failed to work. 

I'm using an official Sony DualShock 3 with original USB charging cable. It appears under Devices and Printers properly as PLAYSTATION(R) 3 Controller, but if you select it and try to calibrate its buttons, no response from any button input.

I have ScpToolkit installed to the default directory to be safe (before I'd tried it in a custom one), but no luck there either. 

To the point, however, the error message produced is: 
FATAL - Virtual Bus Driver pre-installation failed with Win32 error 2147549183. 

With Scarlet's old drivers the primary issue going into Win10 was that the service wouldn't start, producing an error of 1068, but I didn't have much luck uncovering the source of that issue either, even with some old troubleshooting tips of their own. 

In case it helps:
Running Windows 10 64-bit version.
Have 360 controller drivers installed (from running a wireless 360 controller with other games).
Have tried ScpCleanWipe a few times. 
Have tried rebooting PC a few times.
Have tried resetting controller.
Have tried Force Driver Installation.
Have tried running DriverInstaller as Admin and in compatibility mode with Win8.

nefarius, 2016-01-27:
Try [`sfc /scannow`](https://support.microsoft.com/en-us/kb/929833)

GmrLeon, 2016-01-27:
Gave it a try, even in safe mode, and no luck. Glanced over the logs and, unless I'm overlooking something, didn't see anything that appeared to be associated with this.

GmrLeon, 2016-02-16:
So I finally figured out what the issue was, or at least, an indirect contributor to it. At some point, and for some very dumb reason, I set my System32 folder to Read Only. I'm not even sure why now, or what I was thinking (maybe thinking I was restoring default settings?), but I applied the equally dumb solution of setting it and its contents to no longer be read only.

Fortunately it resolved a number of issues I was encountering with various other bits and bobs of software. The only downside is, having been such a wide open fix, I'm not sure which folder probably should have been focused on to sort this out, as I'm pretty sure certain folders are supposed to be Read Only in Sys32.

closed 2016-07-17 by nefarius

Crossoni, 2017-12-12:
Hey, if someone else is having this problem, I found a solution what worked for me. The problem was that I didn't have Net Framework 3.5 installed and I was unable to install it. After I installed it everything worked perfectly. Here is the guide if you can't install Net Framework 3.5:

Run these commands in Command Prompt (Admin).
a. PressWindows key +X. 
b. Click oncommand Prompt (Administrator).
On the open Command Prompt window copy and paste the commands –
net stop wuauserv
net stop cryptSvc
net stop bits
net stop msiserver
ren C:\Windows\SoftwareDistribution SoftwareDistribution.old
ren C:\Windows\System32\catroot2 catroot2.old
net start wuauserv
net start cryptSvc
net start bits
net start msiserver
pause 
c. Wait for the process to complete.
d. Close the Command Prompt.
Now, try to install the updates. Check if it helps.