nefarius/ScpToolkit #185: bluetooth module can not wake on after sleepmode on laptop

D-Zen, 2016-01-27:
I have a asus K43SJ laptop with bluetooth module, i install driver for buetooth module and ds3 gamepad, it's working well but when my laptop wake on from sleep mode, bluetooth module not wake on and not connect with ds3 gamepad, i used ScpToolkit 1.6.238.16010. How to fix it?

nefarius, 2016-02-06:
What manufacturer is your Bluetooth host? Did you check against the compatibility list?

googleygoggles, 2017-06-18:
I did the research for him and it appears that his laptop uses an Intel Wifi/Bluetooth Chip. I'm having the same issue with my own Intel AC 8260 WiFi/Bluetooth chip. SCPToolkit doesn't work after waking and the PS3 controller just has the 4 flashing lights. In the SPC Monitor (legacy) the Host Address just says <error>.

Oh I almost forgot, VID_8087&PID_0A2B

closed 2018-01-26 by nefarius