nefarius/ScpToolkit #187: Clicking left stick sometimes registered as Dpad move on Dualshock 3.

KainXVIII, 2016-01-27:
Happens me everytime in Dragon's Dogma PC.

mirhl, 2016-01-27:
Duplicate of #181

dkhelwig, 2016-01-31:
@mirhl  This isn't a duplicate of #181. This issue seems to be related to the sensitivity of 'Digital to Analog buttons'. On DS3 controller all of the buttons have a pressure sensitivity and the threshold value for being pressed seems low to me. 

I'm having this exact same issue but with only one of my two controllers on either of my computers. Windows 7x64 and Windows 10x64. Issue occurs both wired and over bluetooth (one is a Toshiba laptop w/bt the other an ASUS mini bt400 device). Often while playing MGS5 pressing the Left Stick in to sprint results in dpad direction hits. 

I can simply shake my older controller and it will count as button inputs. Occurs both in games and in the Test window for Game Controllers. The newer controller does not have this issue. My PlayStation 3 does NOT register shaking the old controller as input.

mirhl, 2016-01-31:
So.. it's a hardware issue?
Can't you just like set a bigger deadzone?

dkhelwig, 2016-01-31:
Unable to set a higher deadzone on the dpad buttons through the windows
control panels. Currently scp only supports deadzone for sticks.

closed 2018-01-26 by nefarius