nefarius/ScpToolkit #188: Cant use bluetooth keyboard after install

DoeEensGek, 2016-01-27:
Is there any way to still use the bluetooth adapter also for my keyboard?
Also my fake controller wont connect while the real ds3 works fine, any idea?

theczorn, 2016-01-27:
What fake controller are you using? Make/Model/Serial #?

nefarius, 2016-01-27:
@your first question: nope.

DoeEensGek, 2016-01-30:
ok, thx.
@theczorn a p3 controller, VID_054C&PID_0268, dont know where to get more information, It works with cable, only not the ?thumbs? and not with bleutooth at all.

DoeEensGek, 2016-01-31:
Ok update; this version worked better:
https://github.com/nefarius/ScpToolkit/releases/tag/v1.6.133.15324

Only problem now is the joystick doesnt work for the fake controller. Bluetooth does work for both controllers.

mirhl, 2016-02-06:
You might want to change title to just highlight the second problem.

closed 2018-01-26 by nefarius