nefarius/ScpToolkit #190: "Sony DualShock 4 found" but not functional

WhiteZeroX, 2016-01-28:
I can pair my three DS3 controllers just fine, see them in SCP Monitor normally and use them as XInput controllers. I can plug my DS4 into USB, it shows up and works as XInput as well. But when I try to use the DS4 over Bluetooth, I get the "Sony DualShock 4 found" notification but it dosen't work at all otherwise. Here's my logs and screenshot from SCP Server. And yes, I'm normally running SCPService, I only ran the SCP Server app for debug purposes once. 

[ScpToolkit_Log-Package_Nick_1453654623.zip](https://github.com/nefarius/ScpToolkit/files/102437/ScpToolkit_Log-Package_Nick_1453654623.zip)

Note: Issue #180 was very similar to this, though the creator of that issue closed it because he had a bad Bluetooth dongle.

IlianBurgos, 2016-02-09:
Hi, did u find a solution? I got the same problem here

WhiteZeroX, 2016-02-09:
@IlianBurgos Yes I'm still having this issue. Haven't tried working it out since I opened my ticket, not sure what else to try. Possibly removing the DS3 driver and only installing the DS4 one possibly? But I use my DS3 controllers way more often, so I'm waiting it out for a solution.

IlianBurgos, 2016-02-11:
I just read that only bluetooth dongle 2.1 and next versions are supported for ds4, you could check your dongle, that was my case I have a dongle 2.0, works fine with ds3 but only wired for ds4, if you get an answer from nefarius please let me know.

WhiteZeroX, 2016-02-11:
@IlianBurgos yeah, I have a newer model with BT 4.0. I don't think we'd even get the "Sony DualShock 4 found" message if the dongle wasn't able to see the controller due to incompatible BT standards..

IlianBurgos, 2016-02-11:
When i disconect the usb cable from my ds4 the light blink 2 o 3 times and the "Sony DualShock 4 found" message appears, the light instantly turns off, could be a ds4 issue too, did you try another ds4?

WhiteZeroX, 2016-02-11:
Nah I only have one DS4. And I've had it working with the same dongle awhile back when I tired InputMapper.

Jan200101, 2016-03-04:
Try these things they work for a lot of people:

- Check if wire is okay
s

WhiteZeroX, 2016-03-04:
@Jan200101 I do have XPadder installed, but as long as it's not running it shouldn't cause a conflict, I assume. Other than that, my cable is fine and I've re-installed several times, include once ONLY using the DS4 controller.

Jan200101, 2016-03-04:
XPadder isnt a Virtual Joystick so its fine it just translates the stuff over to another language.
One thing you could try is checking the controller tab and see if it is even seen

KinanEldari, 2016-05-22:
**##Possible Fix##** 
with latest 1.7beta drivers

First I installed all the controller drivers and such as per normal (I have a DS3 and DS4)

After that, I did a clean wipe with the clean wipe utility so the bluetooth dongle would be available again. 

_Then_ I paired the DS4 with my PC through the normal bluetooth utility in windows (so _before_ you claim the dongle with Scp Server) by holding the PS and share buttons until the light bar starts blinking quickly twice before going dark.

And finally, I went through the driver install utility all over again, installing everything (as you did do the clean wipe)

There may be a possible shortcut by simply removing everything scp server related, followed by pairing the controller to the PC like a normal bluetooth device, then installing the Scp server, initializing the controller(s) and the dongle, and everything else. But considering my DS4 works again after several hours of mucking about, I'm sure you'll forgive me for not being so eager to try this out.

Hope this helps anyone else!

ReaLx3m, 2016-09-27:
I had the same problem with my DS4, but i managed to resolve it.
The issue was that the DS4 controller didnt get initialy paired upon installation of the driver.Tried on Win 10 x64, clean install Win 7 x86, clean install Win 10 x86 and in all instances it exibited the same behaviour, so maybe the dev could look into this issue.
I resolved it by opening "ScpToolkit Settings Manager" and under "DualShock 4 Specific" checked "Always re-pair controller on Usb connection" , re-plugged on usb and then unplugged and it connected to bluetooth with no problem. After you get it connected you can uncheck the option.

w1r0x, 2016-10-30:
@ReaLx3m Thank you. That helped. Maybe we should check this option by default?

WhiteZeroX, 2016-11-07:
@ReaLx3m That totally worked! Thanks man. This definitely needs to be a default option.

NLmitchieNL, 2018-01-19:
@ReaLx3m  This worked for me as well. After this it plugs in on Virtual Bus right away on Bluetooth. It just seems Dark Souls III doesn't like the DS4. Haven't tested in other games.

closed 2018-01-26 by nefarius