nefarius/ScpToolkit #191: Re-install driver after Windows 10 upgrades

WilliamFeely, 2016-01-28:
How about making ScpToolkit detect Windows upgrades (such as installing a new build of Windows 10) and automatically re-install the Scp driver?  I am currently on Insider Preview, so upgrades happen often, and every time I have to manually re-install the Scp Bluetooth driver because Windows Setup re-installs the generic driver.

Is it possible for you to detect the driver being replaced by generic, associate such replacement with an increment in the Windows build number, and do a one-shot driver reinstall after such an event?

TheBlueTroll, 2016-02-05:
heyo, did you know, in the "ScpToolkit Settings Manager" under services tab, there is a option called "driver settings". here is a Image
[image]
As you can see, the option already exists, so lets first make sure the options are checked.

tl;dr: Go to "ScpToolkit Settings Manager", then pick "services", and confirm they are ticked( all ticked is fine)

Of course its reliant on the Scp Service to be running after such a update (its a service called Ds3Service, which runs the ScpService.exe)

otherwise, its a already implemented feature.

WilliamFeely, 2016-02-05:
Restarted my system and it fixed the driver.

Do you think you could implement checking when a device is plugged in?  I share a Bluetooth adapter with two computers and it wasn't plugged into the second one when it got the latest Insider Preview build.

TheBlueTroll, 2016-02-05:
(Just a reminder, I'm not the Deveolper, I'm just a user.)

This is something that @nefarius would really have to comment on, as I'm not a dev.

closed 2018-01-26 by nefarius