nefarius/ScpToolkit #192: ScpToolkit Driver Installer doesn't complete

khanv1ct, 2016-01-30:
I just get "a fatal error has occurred. Consult the logs for details."

mclang, 2016-02-05:
I have same problem.

Trying to do fresh install of ScpToolkit v1.6.238.16010 on Windows Pro 64bit.

rkubes, 2016-02-12:
Just as an FYI - I had the same error - the "libwdi.dll" missing.

I was able to get it working though.

What I found was the following. When I first installed it listed 4 pre-reqs (Visual Studio packages). I don't recall what they were. For me it stated that I needed all but the third in the list, it claimed the third was already installed.

I proceeded with the install, which included installing the pre-reqs that were missing. I couldn't get it to start and was getting the above error.

I then uninstalled and tried installing again. This time the pre-reqs were inversed. It stated that I had all but the third (which it previously stated I had). I let it install that third pre-req, and then the install finished without a problem. It has been working fine since.

mclang, 2016-02-12:
Thanks for the hint, now my DS3 works flawlessly.

TheNoobWar, 2016-07-11:
Facing similar issues of it running the entire driver as a 32-bit process on a 64-bit machine. All the pre-requisites are installed. Machine is on a Windows 7 64-bit OS.

`<log4j:event logger="ScpDriverInstaller.MainWindow" timestamp="1468230996794" level="INFO" thread="1"><log4j:message>SCP Driver Installer 1.7.277.16103 [Built: 04/12/2016 08:58:01]</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpDriverInstaller.MainWindow" method="Window_Initialized" file="" line="0" /></log4j:event>
<log4j:event logger="ScpDriverInstaller.MainWindow" timestamp="1468230996819" level="INFO" thread="1"><log4j:message>Microsoft Windows 7 Home Premium (6.1.7601.65536 AMD64) detected</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpDriverInstaller.MainWindow" method="Window_Initialized" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1468230996874" level="DEBUG" thread="1"><log4j:message>Preparing to load libwdi</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1468230996877" level="DEBUG" thread="1"><log4j:message>Called from 32-Bit process</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1468230996877" level="DEBUG" thread="1"><log4j:message>libwdi path: C:\Program Files\Nefarius Software Solutions\ScpToolkit\libwdi\x86\libwdi.dll</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1468230996877" level="FATAL" thread="1"><log4j:message>Couldn't load library C:\Program Files\Nefarius Software Solutions\ScpToolkit\libwdi\x86\libwdi.dll: System.ComponentModel.Win32Exception (0x80004005): The specified module could not be found</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpDriverInstaller.App" timestamp="1468230996879" level="FATAL" thread="1"><log4j:message><![CDATA[An unexpected error occured on application startup: System.DllNotFoundException: Unable to load DLL 'libwdi.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at ScpControl.Driver.WdiWrapper.wdi_create_list(IntPtr& list, wdi_options_create_list& options)
   at ScpControl.Driver.WdiWrapper.get_UsbDeviceList()
   at ScpDriverInstaller.MainWindow.OnUsbDeviceAddedOrRemoved()
   at ScpDriverInstaller.MainWindow.OnSourceInitialized(EventArgs e)
   at System.Windows.Window.CreateSourceWindow(Boolean duringShow)
   at System.Windows.Window.CreateSourceWindowDuringShow()
   at System.Windows.Window.SafeCreateWindowDuringShow()
   at System.Windows.Window.ShowHelper(Object booleanBox)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at System.Windows.Threading.DispatcherOperation.InvokeInSecurityContext(Object state)
[... cut ...]
   at ScpDriverInstaller.App.Main()]]></log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="Shahriar-PC" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:UserName" value="Shahriar-PC\Shahriar" /><log4j:data name="log4net:HostName" value="Shahriar-PC" /></log4j:properties><log4j:locationInfo class="ScpDriverInstaller.MainWindow" method="&lt;.ctor&gt;b__0" file="" line="0" /></log4j:event>
`

cephie3, 2017-04-29:
mine wont finish installing says that cabinet error or corrupt..am on windows 7 64 bit...pls help!!

closed 2018-01-26 by nefarius