nefarius/ScpToolkit #193: Bluetooth not working after install SCP driver

nikoroxy, 2016-01-30:
When I install the SCP driver, bluetooth menu in settings>devices menu disappear and i can't connect any other devices. It restart to work when uninstall SCP toolkit and reinstall Intel AC 7265 driver.

Any solutions?

theczorn, 2016-02-02:
This is expected behaviour. The SCP toolkit will pick a Bluetooth radio to override and sync DS3/DS4 controllers to. From the [original forum post by ScarletCrush](http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller):

> Note : The Bluetooth Driver requires that you dedicate a Bluetooth Dongle to DS3/DS4 use. It will not > be usable by other Bluetooth devices after the driver has been installed.

If you're overriding your main Bluetooth radio you could try to get a supported Bluetooth USB dongle and overwrite that with SCP instead. The DS3/DS4 controllers would then use SCP to sync to your USB dongle, while everything else would go to the main radio.

nikoroxy, 2016-02-03:
Thanks a lot. I'll buy a dongle :)

closed 2016-02-06 by nefarius