nefarius/ScpToolkit #194: DS3 Joysticks work, but no other input will work

nedm89, 2016-01-31:
[ScpToolkit_Log-Package_Bitchbox_1454224699.zip](https://github.com/nefarius/ScpToolkit/files/111162/ScpToolkit_Log-Package_Bitchbox_1454224699.zip)

theczorn, 2016-02-09:
When you say "no other input will work" do you mean:
a.) Only the DS3 joysticks receive input. None of the other controller buttons work.
OR
b.) The DS3 controller works with bluetooth, but all other bluetooth devices (mice, keyboards, etc) become unresponsive.

Is this only occurring when using a USB connector, using bluetooth, or for both?

nedm89, 2016-02-09:
I appreciate your reply, the joysticks wouldn't work but after reinstalling
everything it works fine. Thank you so much for the follow up email though

closed 2018-01-26 by nefarius