nefarius/ScpToolkit #196: Mouse doen't work after install usb/BT

Hactarus, 2016-02-01:
I use logitech wireless keyboard + mouse and official CECHZC2E sony DS3 Controller on Windows 10.
I install drivers for BT and/or USB for this controller, I loss my mouse, and need to unistall drivers with keyboard to fix.
In usb mode crontoller is detect by win10, but with 00:00:00:00:00:00 MAC address (for host and pad1) and no input work.
Do you need more informations ?

Hactarus, 2016-02-01:
No mouse after reboot clean and unistall.
Controller is now right usb detected (but no input yet).
http://up.edmund.ovh/file/igtkt2gbl5BcFn0h/OFMOCpOB3qtF6Irp/ScpToolkit_Log-Package_LanParty_1454366425.zip

mirhl, 2016-02-06:
BT adapter needs dualshock compatible driver. 
It cannot be fixed.

llakssz, 2016-02-16:
DS4Windows works with my DS4 and bluetooth devices still work (speakers etc) - but this is a problem with ScpToolkit? Why?

mirh, 2016-02-16:
Because DS4 has a PC specific mode where it just work with bt own "controller" profile. 
DS3 hasn't this luxury afaik.

llakssz, 2016-02-16:
Ah ok thanks, that makes sense :)

SebBareyre, 2016-03-07:
I have the same problem.. Any help ?

llakssz, 2016-03-08:
I assume you have uninstalled the driver, right?

SebBareyre, 2016-03-08:
Yes but my computer didn't detect my bluetooth mouse

mirh, 2016-03-08:
Open device manager and check?

charlesmilette, 2016-03-08:
Try to restart your computer, for it to reinstall the original driver

Jan200101, 2016-03-08:
Quick question does your mouse etc. connect over a bluetooth dongle or its own dongle ? This may explain somethings

magevideogames, 2016-04-09:
In order to have a bluetooth mouse still work, you must disconnect the bluetooth dongle before installing the controller drivers. Then you can plug it back in after the installation finishes and it will work. That is what I have to do in order to not lose my mouse. You have to do the same thing before starting the service as well. This could easily be solved by allowing the drivers to be installed without bluetooth support. As of the latest version, the bluetooth driver is still installed even if you uncheck it.

mirh, 2016-04-10:
Or you can just open device manager and revert back to default driver. 
Anyway, hopefully it will be fixed in next incoming version.

closed 2018-01-26 by nefarius