nefarius/ScpToolkit #197: Couldn't open device

mmathys, 2016-02-03:
For some reason, Scp isn't working for me.

- Normal installation (DS3 only, no Bluetooth, no DS4), Windows 10, all dependencies installed, latest version

- Previously installed MotionInJoy. Is this a problem?

[image]

Thanks for your help.

SiTWulf, 2016-02-03:
Maybe motionjoy are not completely uninstalled, the error tells you about OPEN DEVICE... maybe the device is in use with another application (Motionjoy) ...
Please double-check the uninstallation of MotionJoy ...

mmathys, 2016-02-03:
Ok so here's what I did in order to uninstall MotionInJoy:

- Uninstall by uninstaller
- Device Manager: show hidden devices and uninstall any device with a MotionInJoy (I think I unistalled like 3 or so)
- I did `pnputil.exe -e` and `pnputil.exe -f -d oem#.inf` the corresponding file (for me `oem1.inf`)
- Deleted all registry files with motioninjoy (1)
- rebooted.

Is there any way I forgot something?

Still, same error.

mmathys, 2016-02-04:
UPDATE: Reset the PC and installed the drivers again.

Still, the same error as above.

While installing, I got this error too:

[image]

mmathys, 2016-02-07:
@nefarius Could a format and reinstalling of windows help?

nefarius, 2016-02-14:
@mmathys hold on 'till the next version, there are some huge changes involved.

axalon, 2016-02-14:
@nefarius When do you think the next version will be released?

LostEnder, 2016-02-19:
Just wanted to note that I am also having this specific issue. I also actually reinstalled Win 10 in this process for other reasons, but no dice.

Here's hoping the new version comes soon!

syepcy, 2016-04-21:
having the same issue :(

nefarius, 2016-04-24:
Please post a screenshot of the Universal Serial Bus Devices section in your Windows Device Manager.

Mezure, 2016-05-06:
I have the exact same issue with one of my dualshock 3, but the driver works perfectly fine with my other dualshock 3.

Windows 10 and I also had motioninjoy previously installed.

[image]

Maxiim112b, 2016-05-29:
Hey ! Just saw this post, I was having the same problem and I fixed it ! It worked for me: 
[image]

When I first installed I didn't checked my controller in the driver installer. I did it and it worked ! Hope it can help you ;) 

Max.

cevinator, 2016-06-30:
Had the same problem, I hit the little reset button on the back of the DS3 once and it's all fixed now. You should try it out.

splurmp, 2016-07-17:
I'm having the same exact problem as OP. I tried the reset button like cevinator suggested but that didn't work. Never had MotionInJoy installed though so it can't be that. 

I did however, have a previous (perfectly working) version of ScpToolkit installed. Unfortunately I was very annoyed by the updater notification that would pop up all the time. I read that it was fixed in a newer version... so I did everything as instructed in this guide: https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.276.16101-BETA

But upon installation I got that same "Couldn't open device" error OP is getting and my DS3 controller didn't work. I tried a bunch of things but no luck and eventually gave up and decided to go back to the old version since that at least worked... but now that doesn't work either. Great.

So now I have no way of using my controller. I'm guessing something messed up with the CleanWipe utility? I dunno. I wish I had just dealt with that annoying notification rather than break everything permanently.

sgtpepper42, 2016-08-10:
Hey I'm also having this problem. Every time I plug my controller in I get that "Couldn't open device" message and whenever I first start the program or try to install the driver for my PS3 controller I get the "System.Nullreference" error message. 

I also used to have MotionJoy, but as far as I can tell I got rid of it. 

Tried resetting my controller, didn't work. Tried redownloading my PS3 driver, didn't work. I'm at a loss now.. any help would be nice.

herberto97, 2016-08-23:
hey is it advisable to use the latest version of scp...i have a lot of problems with it.It never reads the controller.ON the pic below you'll notice that it requests that i connect my controller..but the controller is already plugged at that very moment.this worked very well at the previous version.i only switched to better the service
[image]

Volkanite, 2016-08-23:
@herberto97. I have the same issue with the latest driver installer. The controller must be present in Device Manager under Human Interface Devices. If not, the installer won't see it, as it only enumerates devices under that node by calling SetupDiEnumDeviceInterfaces(GUID_DEVINTERFACE_HID);

I don't know if this is how previous versions of the Driver Installer did it but i think the current version should be fixed to enumerate ALL USB devices, not just those HID compliant devices.

Anyways the workaround for this issue would be to get your controller back under Human Interface Devices. This can be done by manually uninstalling the current driver in Device Manager.

herberto97, 2016-08-25:
@Volkanite  thanks for the help..but that doesnt seem to work..suppossing u wanted me to uninstall it the ps3 drive under "libusb-win32 devices"...but anytime i do so ,it reinstalls and remains there and doesn't go under HIDs..its either i misunderstood you or it doesnt go here....lemme send you a lil pic of when its uninstalled
[image]
...thanks again

Volkanite, 2016-08-28:
Follow the steps below in order to revert to controller to an HID compliant device.
- Right click device
- Update Driver Software
- Browse my computer for driver software
- Let me pick from a list of device drivers on my computer
- USB Input Device
- Click OK

herberto97, 2016-08-28:
@Volkanite  Thank you for that ....but i think i'll stick to the previous version,given that i can't use my in-built bluetooth with my controller in the latest version....Thanks again tho...hope i get to help you in the future

Volkanite, 2016-08-28:
@herberto97 Okay but did my instructions work to get the controller back under Human Interface Devices? The Bluetooth can be fixed as well. The Driver Installer calls SetupDiEnumDeviceInterfaces(GUID_BTHPORT_DEVICE_INTERFACE); which means the device has to present under Bluetooth Radios in Device Manger when you run the installer. This can again be done by reverting to the default Bluetooth driver.

herberto97, 2016-08-28:
@Volkanite yess it did work...my DS3 controller was recognised after it entered HIDs....Pls
 can you throw more emphasis on conveting the bluetooth driver?

Volkanite, 2016-08-28:
@herberto97 Follow the steps below in order to revert to the default bluetooth driver.
- Right click the bluetooth device
- Update Driver Software
- Browse my computer for driver software
- Let me pick from a list of device drivers on my computer
- Generic Bluetooth Adapter
- Click OK

herberto97, 2016-08-28:
@Volkanite Didnt see any Generic Bluetooth Adapter..but here's what  i saw 
[image]
there's my bluetooth driver
[image]

Volkanite, 2016-08-28:
That should do it. So the Bluetooth adapter now shows up in ScpToolkit Driver Installer?

herberto97, 2016-08-28:
Unfortunately it doesnt..it doesn't go under HID nor is it recognised by scp

Volkanite, 2016-08-29:
@herberto97  Can you run this small piece of code and tell me what the output is?
https://github.com/Volkanite/bluetoothsearch/releases/download/r1/bluetoothSearch.r1.zip

Run it from command prompt.

bizaff, 2016-09-03:
So back to mmathys's original problem - which is the same as my problem except I've never installed MotionInJoy on this machine.  I've seen other issues talking about code signing and what not.. what's the latest?

UORasputin, 2016-09-11:
I found my DS3 Driver in the Device Manager under a strange category and uninstalled it, removing the driver.
After that I could finally install the current driver using the Driver Installer.
It is now listed correctly under USB Devices.
I'm using Windows 10 and had MotionJoy and the old "latest" version installed (which worked).

I think the category was something like libusbk.

bardozan, 2016-09-14:
If there is anyone having the same problem, I can confirm that resetting the DS3 via pushing the little button at the back side works.
Thank you for sharing it.

bizaff, 2016-09-17:
@bardozan - you're absolutely right - that fixed it for me.  This should probably be in the install instructions.

andresitogp, 2016-09-18:
Hi Guys, so, I installed SCP's windows service, bluetooth driver (making sure I tick the built-in bluetooth adapter for my laptop), and dualshock driver (making sure I tick the PS3 controller). It works then, but when I restart I loose my keyboard and one of my usb hard drives. I'm able to login using the on-screen keyboard, uninstall SCP drivers, and the the keyboard and HDD come back immediately. Am I doing something wrong? I have two usb HDDs, but only one dissapears.

meatathon, 2016-09-21:
The solution from @bardozan worked for me too.  Two of my controllers worked, and the other two did not.  Once I pressed the button on the backs of the two with the problem, all four were installed.

fevat4574, 2016-10-16:
bardozan i reset button idea worked for me. ty so much

Deltomane, 2016-11-30:
Hi guys, same problem here!
As I wrote here https://github.com/nefarius/ScpToolkit/issues/110, I'm using my DS3 fake controller via usb cable.
It works for a few seconds, but as soon as I press a button the contoller seems stuck.
When I open the SCPserver window, it shows the same message "Couldn't open device".
I tried the reset button but nothing changed.

Master-Kay, 2016-12-12:
Ok, so... I´m having the same original discussed problem, that mmathys had, he probably solved it with the new updates, good for him, I recently formatted my PC, after reinstalling all browsers and essentials, I went for the scp, as usual, installed the drivers and all, with my controller connected, and it just doesn't open it, the software recognizes the DS3, but it doesn't open the device, I tried everything possible in my hands, reseting the controller, reinstalling the drivers a hundred times, I even tried to use the original scp, it even worked, but after one day it happens the same thing, it recognizes the device, but it doesn't open. 
I really hope someone could help me, this is very important for me, very much important.

Thanks for now

gabrielbarr0s, 2017-01-10:
Resetting the DS3 by pushing the little button at the back side worked for me.

Ctesias, 2017-02-14:
Just ran in to this issue trying to install my DS3 controller.   My problem was that I had the controller plugged in through a USB hub and not directly attached to my tower.

Moving the plug directly to the tower and choosing no specific device in the dropdown to the right when I installed the drivers cleared it up for me.

As an interesting aside, I was sharing the hub with the PS3 controller and my USB blueooth headset.  Whenever I install the DS3 drivers with the tool, it kills all the functionality on my headset aside from the sound.

Bre3zeR, 2017-03-18:
@gabrielbarr0s @nefarius After searching for 4 and half an hour, watching countless of guide, troubleshoot. THIS is what fixed my issue about the open device issue (controller showing as 00:00:00:00:00) and SCP telling me the error.

Since it cost nothing to reset the controller with the little button being it, i advice anybody having issue doing it.

Thanks you sir, you just made my day that much better.

chemitasama, 2017-05-24:
Hi!
I have the same problem, i think so...

Worked correctly but suddenly stopped doing it. I get error "could not open device ..." and "virtual bus error", I tried to uninstall scp, reinstall, restart, stop services, but I can not find solution.
:(

thx for all. if doesn't work thx for all time i can play with my ds3
[image]

closed 2018-01-26 by nefarius