nefarius/ScpToolkit #200: BTH DS3 Device

Clixtons, 2016-02-07:
all my dongles are automatically converted to bth ds3 device, how can i remove so it wont to that.
i only want one dongle for my controller and other dongles for other devices.

mirhl, 2016-02-07:
If you are sure you haven't ticked bluetooth devices to overwrite in the installer then I guess this could be considered a duplicare of #174 
Or better to wait for nefarious to know if selection of *every* bt dongle installed shouldn't on the other hand be the point here.

theczorn, 2016-02-09:
What device(s) are you installing the Bluetooth drivers on in the Driver Installer? IIRC it's possible to target a USB Hub and overwrite all your dongles at once when you intend to target a specific USB dongle.

mirh, 2016-02-17:
#218 seems to confirm my second guess.

closed 2018-01-26 by nefarius