nefarius/ScpToolkit #202: DS3 keeps disconnecting and connecting again every 30 seconds.

IfritTower, 2016-02-07:
I have uninstalled and reinstalled multiple times to try and solve this issue. The DS3 works fine when it is active, but it disconnecting and almost immediately reconnecting around every 30 seconds, leaving me with a few seconds without control. The issue has started happening when I use the DS3 with a normal PS3 too, which it did not use to before I installed the wrapper on my PC. It seems almost like the issue waits until there haven't been any inputs for a few seconds before popping up, as when I am in the middle of a long chain of inputs (like in a combo in a fighting game), it doesn't appear until stop. My debug log is below. Thanks for your help!
[ScpToolkit_Log-Package_Juan_1454884007.zip](https://github.com/nefarius/ScpToolkit/files/120959/ScpToolkit_Log-Package_Juan_1454884007.zip)

vorpalMachine, 2016-04-11:
Same issue here. Driver 1.6.238.16010, Windows 10. If I knew how to include a debug log, I would.
For me the amount of time varies from 30 seconds to a few minutes, but it will simply disconnect and reconnect. Not fun when playing Dark Souls!

charlesmilette, 2016-04-11:
[image]

closed 2018-01-26 by nefarius