nefarius/ScpToolkit #206: Error 0x80004005 while connecting via bluetooth

remielowik, 2016-02-09:
While im connecting using the cable it all works fine but when i disconnect to connect using bluetooth it gives the 0x80004005 errorcode(using the old program i was able to do both the bluetooth and the usb). 

Logs
[ScpDsxServer_Remie Lowik.log.xml.txt](https://github.com/nefarius/ScpToolkit/files/122988/ScpDsxServer_Remie.Lowik.log.xml.txt)

theczorn, 2016-02-11:
- Have you tried updating your Bluetooth drivers?
- What model controllers are you trying to sync?
- What OS version?

This appears to be a generic Bluetooth error message in regards to pairing, more details would be helpful.

Jan200101, 2016-03-04:
#224 If have put some things in there if you know how to fix it yourself now close the thread if you dont tell me

closed 2018-01-26 by nefarius