nefarius/ScpToolkit #207: problem changing assigned control number

wiloh95, 2016-02-09:
Hi! I have 2 original dualshock 3 controllers and 2 original xbox360 controllers for my pc (both used wirelessly). Currently the ps3 controllers are number 1 & 2 and the xbox360 controllers are 3 & 4. I want to change this around because I primarily want to use my xbox controllers. 

I have checked the box under "service settings" which is called "Skip occupied gamepad slots". But disconnecting all controllers and then reconnecting the xbox controllers still make then number 3 & 4. 

So what is the proper way of changing the controllers around so that I can use my xbox controllers as my first choice? Any help would be appreciated :)

And thanks for a wonderful software, I'd donate if I wasn't so poor....

mirhl, 2016-02-09:
By design, I believe XInput is meant to assign controller slot after connection order. 
If you want to arbitrarily chose that, I believe that should be logically better done in [x360ce](https://github.com/x360ce/x360ce)

nefarius, 2016-03-14:
This behavior will be fixed in the upcoming release; then the DualShock`s will behave just like Xbox controllers and will respect connection order, free slots and slots occupied by real Xbox controllers.

closed 2016-04-10 by nefarius

SneakyLittleman, 2016-07-04:
I'm using driver 1.7, a bluetooth DS3 controller and a wireless Xbox One controller. Regardless of connection order, the DS3 pad is always assigned as pad number one and Xbox controller as pad number two (tested in PES2016 and Portal 2).
The DS3 pad has its "1" led on. Any way to fix that? Thanks.

mirh, 2016-07-04:
Did you try the always unplug option in the settings?

SneakyLittleman, 2016-07-04:
Yes, the setting is ticked.
After a few tests, here's what happens, from a cold boot (win 10 x64 all updated):
- I open "set up usb controllers" 
- switch on the Xbox one controller with the xbox button, and the pad is then listed in the "game controllers" window
- switch on the DS3 bluetooth pad with the Playstation button, and the pad then appears in the "game controllers" window as well.
- launch a game (PES2016) - only the DS3 pad can navigate the menus (primary controller). When launching a match with two players, Xbox one controller is clearly the second controller / pad 2 or whatever you want to call it.

Now, on occasion (twice, actually), I've managed to have the Xbox one pad as the primary one, but I have no idea how...and I have no idea if it's a bug or a setting in scp...

agodinhost, 2018-01-21:
I could not  figure out how to make SCP to work in Game Mode on windows 10.
Anyone?