nefarius/ScpToolkit #212: Webcam recognized as bluetooth dongle

charlesmilette, 2016-02-14:
Look at the screen:  
[image]

HP Truevision HD is my webcam...

BTW why is the PS3 controller listed here ? It isn't a bluetooth dongle

nefarius, 2016-02-14:
There is no check **what** the listed devices are, that's why *you* - the user - have to tell the program what's a Bluetooth dongle. The list just shows all detected USB devices.

closed 2016-02-14 by nefarius

charlesmilette, 2016-02-14:
Wow, that was fast.

nefarius, 2016-02-14:
I'm currently online ;)

charlesmilette, 2016-02-14:
If I install the bluetooth driver, will I be able to still use it for other things, because I only have the integrated bluetooth chip, and no dongle at disposition.

nefarius, 2016-02-14:
**No** you won't, the Dongle becomes dedicated to SCP and all currently connected Bluetooth devices will immediately lose connection.

charlesmilette, 2016-02-14:
Any whay to pair my DS3 with the computer without a special driver?

nefarius, 2016-02-14:
Nope.

charlesmilette, 2016-02-14:
;(

charlesmilette, 2016-02-14:
Finally, i said "Screw this" and I overwrote my bluetooth driver. I never used it before. BTW, how can I disable bluetooth (like not active) when my laptop is on battery, I need the more battery life possible for school.

nefarius, 2016-02-14:
Do you have a dedicated switch one one of your laptops side? Typically you can turn off Bluetooth and wireless networking with that. If not - sorry, no power savings implemented in the software.

charlesmilette, 2016-02-14:
No I don't have a switch on it, but I have a airplane mode button on it (FN + F12) I think this button is implemented with software, not hardware, so it will not work. (When I resetted my PC, I had to install a driver this button to work)

charlesmilette, 2016-02-14:
I just tried, it doesn't works. It just disables the Wi-Fi via Windows

nefarius, 2016-02-14:
The software-turn-off typically only works with the original manufacturers drivers.

charlesmilette, 2016-02-14:
If I stop and start the service, it then saids "Initializing Bluetooth host", does that means that stopping the service will stop the Bluetooth?

nefarius, 2016-02-14:
Well, what's your expectation on "stop the Bluetooth"?

charlesmilette, 2016-02-14:
Full disable, like when you turn the Bluetooth switch on or off on a cellphone.  
I can't just unplug the dongle, the card is inside my computer. Plus the same card also gaves me WiFi.