nefarius/ScpToolkit #214: DS3 extraneous button presses being generated

illarionds, 2016-02-15:
(Sorry, posted this first on http://forums.pcsx2.net/Thread-ScpToolkit-XInput-Wrapper-aka-ScpServer-Reloaded before noticing the request to put issues here)
-----------------

This program is *so* much better than any alternative I've found, thank you!

However, I'm having a problem on my main machine.  It seems like random button presses are being generated - for example, playing Arkham Knight, I find myself randomly ejecting from the batmobile (L1), turning on detective mode (d-pad up), and bringing up the gadget selector (d-pad down), without pressing those buttons.

In Binding of Isaac, I accidentally drop bombs (L1).

In both cases, it only happens while other buttons are being pressed - I wouldn't, for example, drop a bomb if I just left myself standing still.  I haven't narrowed it down to any specific combination that triggers the problem though, nor is it reliably repeatable.  

I can't see anything wrong in Xinput tester or the Windows gamepad calibration utility - but the problem is intermittent, I'm not sure I necessarily would see anything.

I'm using a DS3 pad, wired, Win7 64.

I have tried a different DS3 pad (both official, both perfectly working with PS3) and USB cable, to no effect.

I'm very very willing to give any debugging info I can - currently this makes the program unusable for me, at least for any realtime/action game.

Debug info attached,
[ScpToolkit_Log-Package_apg_1455535826.zip](https://github.com/nefarius/ScpToolkit/files/130598/ScpToolkit_Log-Package_apg_1455535826.zip)

Many thanks for any help you can give!
[Uploading ScpToolkit_Log-Package_apg_1455535826.zip…]()

Ajama11, 2016-03-12:
I had the same problem with my controller, and the issue was the controller itself. From wear and tear of using it for many years, whenever I moved the left analog stick a button would fire for a split second (typically L1 or the Dpad). I looked online and many other people ran into this problem from just old controllers. I bought a new controller, and the problem went away.

SlowSeer, 2016-03-13:
This is also happening with one of my controllers. When looking at the 360 button layout under game controllers, X and R1 are being pressed frequently on their own, but when I check Input Tester in SCP Monitor the buttons look like they are fine.

It's possible that my controller could be registering minuscule amounts of false pressure on the buttons that aren't visible on Input Tester since it has a sliding display, where the 360 layout is on/off. 

It would be awesome to have a pressure sensitivity dead-zone for normal buttons too.

closed 2018-01-26 by nefarius