nefarius/ScpToolkit #216: When installing, unchecking the tray app result in a false error message.

charlesmilette, 2016-02-16:
When I install with the tray app unchecked, I get the error "File ScpTrayNotifications.exe not marked for install." Or something like that, I don't recall correctly. It then says the setup failed, even if everything succeded before. Actually, everything is okay (The start menu links and program files are here, as the uninstall entry.).

Jan200101, 2016-03-02:
Uninstall it reinstall it and disable the tray app from autostart

charlesmilette, 2016-03-02:
Actually, I just kept the install like that and I got nothing wrong. The error is just something to get rid of, because it is not true it failed. All succeded, except the tray aoo.

Jan200101, 2016-03-04:
Then nothing is wrong everything is okay the error is just there because it isnt marked (like the error says) lets quickly put @nefarius in here so he knows

closed 2016-04-10 by nefarius