nefarius/ScpToolkit #217: Using Windows Notifications instead of the custom ones?

llakssz, 2016-02-16:
Hi, on Windows 10 (maybe 8 too) - there is a notifications system built in and many apps use it (including DS4Windows).
But scptoolkit uses it's custom notifications and it's a bit intrusive I think. I can disable the notifications sure, but notifications are good, but it would be much better if the windows notifications were supported/used instead.
Thanks :)

laxu, 2016-03-07:
I agree. Not only does SCP's own notification system look like ass, it also tends to fire excessive amounts of notifications. There is really no good reason why it should not use the stock notification system.

charlesmilette, 2016-03-07:
The notifications could also use the old OS' toast notifications when not in the Win10 or Win8 notifications center

nefarius, 2016-03-14:
Weeeeeeeeell, yeah, it was slapped in rather sloppy I have to admit ;) My main motivation of introducing them was so users would know whats going on in more detail without having to consult the logs with a separate log viewer all the time. I agree there is room for improvements though.

closed 2018-01-26 by nefarius