nefarius/ScpToolkit #219: Buttons not working ingame

canadia1, 2016-02-17:
Okay so I installed the program and the drivers just as described. I find my dualshock 3 controller and it seems to be working in the monitor thing. But when i launch a game (Mad max, Lords of  the fallen, Need for speed etc) none of the buttons work in game. I recently used Motioninjoy but after so many bluescreens I got this instead. I removed all the drivers and files from Motioninjoy. Also my controller is a legit one. 
[ScpToolkit_Log-Package_Nikke_1455648192.zip](https://github.com/nefarius/ScpToolkit/files/134102/ScpToolkit_Log-Package_Nikke_1455648192.zip)

mirh, 2016-02-17:
Make sure to *[really](http://pcgamingwiki.com/wiki/Controller:DualShock_3#Removing_MotionInJoy)* have uninstalled it.

canadia1, 2016-02-17:
I have. I uninstalled it and removed any files that came with the program. including the douchy driver file that causes BSOD. Right now I'm getting BSOD from the SCPToolkit. It seems that it's the scpvbus.sys file that's causing it.

canadia1, 2016-02-17:
Okay so I did a complete re-install of windows and now it's working. Except for that after i have been playing for a while (like 5-10 min) the controller stops working. And so does my sound device apparently (it's an USB connected headset). Unplugging and plugging it back in doesn't work. The only way is to restart my computer. Which is quite annoying s
[ScpToolkit_Log-Package_Nikke_1455737382.zip](https://github.com/nefarius/ScpToolkit/files/135021/ScpToolkit_Log-Package_Nikke_1455737382.zip)
ince this happends after like 10 min of playing.

mirh, 2016-02-17:
Check if Ds3service is still running.

canadia1, 2016-02-18:
SCP DSx Service is running.

kallkalony, 2016-02-25:
I am also having the same issue. Everything registers on the input tester, go to use it in game and it doesn't even exist. Aside from that I also have to install the drivers every time I start my computer for it to even recognize the controller and give me the option to test its input. Some help would be much appreciated.

closed 2018-01-26 by nefarius