nefarius/ScpToolkit #220: Input not recognized by games

yrro, 2016-02-17:
On a Windows 7 system with a DS3 controller, the input tester shows the controller buttons & axes working but when running Dirt 3 the controller inputs do nothing.

This is similar to issue #219, but I didn't want to hijack some else's bug.

Here's the output from the debug tool: [ScpToolkit_Log-Package_Pete_1455719562.zip](https://github.com/nefarius/ScpToolkit/files/134491/ScpToolkit_Log-Package_Pete_1455719562.zip)

mirh, 2016-02-17:
> This is similar to issue #219, but I didn't want to hijack some else's bug.

You don't seem to grasp the idea of bug report. This isn't a "forum". You *are* expected to "hijack" bugs (which aren't "owned" by reporters btw ;))

Jan200101, 2016-03-02:
Do you have something like Vjoy installed ? It breaks some games e.g. The Binding of Isaac and others

closed 2018-01-26 by nefarius