nefarius/ScpToolkit #221: DS3 R2 L2 not working

Ludancv, 2016-02-17:
I've seena couple of threats about this issue but I want to provide as much info as possible.

My controller is a DS3 original, it works perfectly when connected with the USB cable, while using the bluetooth the triggers (L2 R2) dont work at all, the thing is that the profile manager does detect L2 and R2 when I press them, I've tried to remap them to keyboard actions and nothing, I also tried to use Joytokey for that matter with no result. The calibration tool doesnt work neither.

My bluetooth adapter is a BT-8001-BK it says it supports V2.0 with 3mbps data rate.

There must be a way to turn the signal that shows in the profile manager into button presses, I hope someone can help me.

Just as a final note, I cant use the other options of the profile manager, I only get the 
button mappings tap active the others are not working wherever I click.

Thank you for your time.

hikkikuma1991, 2016-04-24:
I have this problem too.

MrRobot2211, 2017-05-21:
same issue

closed 2018-01-26 by nefarius