nefarius/ScpToolkit #222: Dualshock 3 not working whatsoever

Natusake, 2016-02-18:
The controller lights do not turn on, and no inputs are registered at all. Here is what I have done:

Step 1: Install ScpToolkit
Step 2: Start ScpToolkit Driver Installer with Administrator Privileges
Step 3: Uncheck Install Bluetooth driver and Install Dualshock 4 driver
Step 4: Check the controller to install the Dualshock 3 driver: 

> PLAYSTATION(R)3 Controller (VID: 054C, PID: 0268)

Step 5: Click Install
Step 6: Open Control Panel, open Devices and Printers, right click PLAYSTATION(R)3 Controller, open Game controller settings
Step 7: Notice that there are no installed game controllers in the menu.
Step 8: Open ScpToolkin Monitor
Step 9: Notice the values given for Pad 1: 

> Host Address : 00:00:00:00:00:00       Pad 1 : DS3 00:00:00:00:00:00 - Usb FFFFFFFF Charging

Step 10: Begin to cry

[ScpToolkit_Log-Package_Nathan_1455756006.zip](https://github.com/nefarius/ScpToolkit/files/135461/ScpToolkit_Log-Package_Nathan_1455756006.zip)

Faane, 2016-02-28:
I had been having the same issue and the only way I got it to work was by running the "SCPServer.exe"

Jan200101, 2016-03-09:
Wait wait wait have you restarted between 8 and 9 ?

closed 2018-01-26 by nefarius