nefarius/ScpToolkit #224: Need to install BT driver after shutdown/hibernation

Ansem93, 2016-02-22:
Hi, I have windows 10 64-bit.
Sometimes windows uninstall automatically BT drivers after shutdown or hibernation, even if I use force install.
There's no way to avoid that? :(

charlesmilette, 2016-02-22:
See bug #51, this is probably your problem. The driver isn't _uninstalled_ it is _not working_.

erapkin, 2016-02-22:
this looks a lot like issue #104

Ansem93, 2016-02-23:
Update about my problem: yesterday I had a blackout and when i turned on my pc, my joystick worked fine even with BT.
Today after hibarnation instead it didn't work, I tried to reinstall BT driver, and still didn't work.
Then I restarted my pc, and BT worked fine without reinstall driver.
I'm really confused @_@

Ansem93, 2016-02-29:
I tryed the guid in the problem #51 ,and left the autmatic install of BT drivers disabled.
After 2 hibernation I still don't have problems, I'll tell you if I still have problems :)

Ansem93, 2016-03-01:
UPDATE:2 days after the procedure, and 3/4 hibernation the joystick is no more working.
Maybe it's because I didn't use it the last time

Jan200101, 2016-03-02:
@Ansem93 do you know if the service is running? 
Here is a quick tool I wrote to check if the service is on. Tell me if it is if it isnt then Ill tell you how to fix.
Sadly GITHUB is weird so I cannot upload it on here so here is a OneDrive Link https://onedrive.live.com/redir?resid=531AF8D5D12A8FC8!2663&authkey=!ADwvHYhVHwuE1IU&ithint=file%2crar

Ansem93, 2016-03-02:
i'll do it asa i'll have this problem again :)

Jan200101, 2016-03-02:
No no no this isnt a fix for that its a tool that just tells you if the Service has problems starting or not

Ansem93, 2016-03-02:
the service is running now, but now the joystick is working. I want to check if the service is running even when i have problem

Ansem93, 2016-03-04:
Ok, today after hibernation the joystick stopped working even with cable, so I used this checker. Here's the results:
- the service was off. when the checker started it, the joystick started work
- if I unplug the joystick the service will go off istantly, so I have to use the checker again
- Joystick is still not working via BT even after using the checker

Jan200101, 2016-03-04:
no no no no thats one thing you are getting wrong the service isnt starting when  you unplug your controller normaly it starts with your pc and then uses your bluetooth dongle to connect to the controller.
THIS CHECKER DOES NOT FIX ANYTHING

Now to fix this open your services (Search Services in the search bar) and fins DS3Service.
After then right click it then go on properties.
After that set it so Late start.
TA DA FIXED now restart your pc

Ansem93, 2016-03-04:
Ok maybe I have find the problem: The service isn't in the list of the windows services, I can only find it into the running serviced in "task menager", but if I search it into the services it's not there.
[http://i.imgur.com/cEirwcQ.jpg](url)

Jan200101, 2016-03-04:
Okay looks like i made a mistake on some Computers it doesnt show up as the service name so look for SCP DSx Service

Ansem93, 2016-03-04:
ok, found it. I set it with delay start :) let's see if this works!

didactic33, 2016-03-04:
I've had similar problem with having to reinstall service from time to time like bug #51 or #104 or this #224 thread. Win7 x64:
[ScpToolkit_Log-Package_Diadactic33_1457114034.zip](https://github.com/nefarius/ScpToolkit/files/159164/ScpToolkit_Log-Package_Diadactic33_1457114034.zip)

 Had it working with a genuine PS3 and a fake PS3 by BT or USB however eventually it fails but I've reinstalled fresh at least 5 times in last month. Now, even with the SCPcleanwipe things aren't working yet this time around.  I used utility DS3service checker and it said "DS3Service was already on and couldn't be turned on, if you see this message tell Jan200101 about it" I ran the analyzer and it generated a report.  I will attempt to attach relevant logs and reports, I apologize in advance since my first time using forum and may not include what you need (attaching log was rejected by GITHUB b/c xml format) but if you let me know what you want to see, I'd be happy to share whatever may help. More details like a hardware report I could share but I will keep it brief until I know what you need.  Many thanks to Jan200101 and all who help make this work!

Jan200101, 2016-03-04:
Okay it seems the Service isnt broken for you I think it may be that there is something wrong :/ sadly I have no idea on how to read the log 100% only about 40% so you have to wait until @nefarius sees this

closed 2018-01-26 by nefarius