nefarius/ScpToolkit #225: ScpProfiler.exe 'Oh Sh... A fatal error occured. Consult the logs for details.'

Khoddah, 2016-02-26:
First of all, thank you nefarius for this great project.

I have a problem with the Profiler, when i open it i get this message 
https://i.imgur.com/NhWH2kV.png

After that i get that it stopped working with the following details
https://i.imgur.com/imyvgbv.png

when i click debug, Visual Studio gives me the following
https://i.imgur.com/gH5w79e.png

And when i check the logs folder, there's nothing about the profiler only scpserver, monitor and installer logs.

Also i ran the debug program included with the tools and couldn't attach it to the post so here it is
http://ge.tt/4ZpxrLY2

The installation directory is: D:\Program Files\ScpToolkit ... if the code has some directory restrictions

OS: Windows 7 SP1
Using Bluetooth: no
Connecting via USB 2 

All i'm trying to do is finding a way to swap the triggers and bumpers for some games (it was through ScpMonitor in older versions) .

Hope that helps. Thanks

sycoforbidden, 2016-03-31:
I have the same problem it seems on windows 10, I guess my entry should be removed?

TheZnayx, 2016-09-02:
I get the "Oh sh..." error, but it doesn't give the window saying that it stopped working after. It (the profiler) just doesn't start/open at all. And all the logs are is a detailing of the installation process - which went without a hitch.

nefarius, 2016-09-04:
You don't have a problem _with_ the profiler, the profiler _is_ a problem... Don't use it. It's not finished and shouldn't have been released in the first place.

closed 2016-09-04 by nefarius