nefarius/ScpToolkit #227: Genuine PS3 controller only works with cable

KlausConrado, 2016-02-28:
Hi.
The controller works fine with the USB cable, but when i disconnect it, the lights of the controller just flash for a while and then nothing happens. Tried to push the hard reset button of the controller, but no luck.
Here is a screenshot of the server log.

Thanks.

[image]

bdefore, 2016-03-04:
I'm getting the same behavior. It never seems to pair with my Rocketfish Bluetooth adapter, which is listed as supported. Happy to have it working while cabled though, so thanks!

bdefore, 2016-03-04:
Disregard, Rocketfish worked correctly for me when I scrolled down to select it as the Bluetooth adapter during install.

Jan200101, 2016-03-04:
#224 Download my Service Checker from there and if it says its off do the steps on my last post.
If you dont get what i mean reply

KlausConrado, 2016-03-05:
Actually i prefer to use the standalone server (ScpServer.exe) instead of the service, this way the service is always stopped.
But i have good news, i tested again today and "magically " the controller is working on bluetooth. Don't know what happens, cause i restarted the computer several times before posting this issue.
The Windows 10 updated itself yesterday, that's the only thing i can imagina that could affect something.

Thanks anyway!

KlausConrado, 2016-03-07:
Just had the same issue with another controller. If you connect a new controller through USB and use only the standalone server (not the service), it will not work on bluetooth if you disconnect the USB. 
However, i figured out how to fix it. Just stop the server, connect the controlller through USB and start the service. Then, the controller will start to work on bluetooth and you can stop the service and open the server again.

**So i guess the problem still exists in the standalone server, cause you have to use the service in the first time for new controllers.**

Oh, and a hard reset (hidden button in the back of the controller) may be needed before this process to make them work on bluetooth.

Thanks.

mirh, 2016-03-11:
Which standalone server are we talking about? I thought service was a required component.

KlausConrado, 2016-03-11:
@mirh It's a executable called "ScpServer.exe". Actually you can't run this server if the service is already running. It's very useful if you don't use the controllers too often and don't want the service running all the time.

mirh, 2016-03-11:
I'm not sure that's expect to work.. 
If I had to judge from the icon, I'd say it's a original scarlett.crush driver leftover.

closed 2018-01-26 by nefarius