nefarius/ScpToolkit #229: DS3 notworking

Hunlight, 2016-03-04:
So I used to use MotioninJoy but peopel suggest  scpToolkit isntead I installed it then installed drivers.Now I went to Game controllers and it shows xbox 360 for windows I click test and its not working buttons/sticks etc. nothing. I read few thread the solution can be to run scpserver ,but when I run it ir always stopped working

Jan200101, 2016-03-04:
1.MotionJoy is not a good program and can harm your system.
2.scpserver is no the right thing you want it just shows if you controller is working based on what you said it should work correct
3.Try uninstalling the controller drivers and reinstall them (Uninstall the XBOX controller and reinstall this tool)

Hunlight, 2016-03-04:
I installed scptoolkit and then it start showing xbox360 before it said smth like game controller etc.

beavermml, 2016-03-06:
i think u should uninstall MotionInJoy first before attempting to install SCPToolkit

Hunlight, 2016-03-06:
I have uninstalled it before done scptoolkit

Jan200101, 2016-03-08:
It can be that your controller is set to the value 0 and that it doesnt update that value sadly I dont know if this glitch can be / has been fixed

closed 2018-01-26 by nefarius