nefarius/ScpToolkit #234: PS Move Controller is outside the scope of this application

gjveron, 2016-03-09:
I just want to make sure that it is documented that the PS Move Controller is outside the scope of this application and it will never work (the one with the color led ball at the top)

The PS Navigation Controller instead, works perfectly (the one with the analog stick)

can anyone confirm this?

charlesmilette, 2016-03-09:
This driver is only for  
[image] and [image]

Their unofficial alternatives may be supported.

Jan200101, 2016-03-09:
It may be added some time please close this Issue

closed 2016-03-09 by gjveron