nefarius/ScpToolkit #235: Update check notification window steals focus

teeedubb, 2016-03-12:
Hi all,
After updating SCP I noticed there is now a automatic check for updates - Very nice, thanks! The thing is, the update notification window will steal focus from the running program, and can be frustrating if this happens in game,

Is it possible to use the same notifications for the update check as the connect/disconnect notifications (the ones that appear on the bottom right of the screen in windows 10)?

nikhosagala, 2016-03-12:
No, you can not. It's a schedule task, you can disable it from the schedule task.

charlesmilette, 2016-03-12:
@nikhosagala A simple change in the code could do it. The only problem would be backward compatibility.

nefarius, 2016-03-12:
All future releases will suppress this window on the scheduled checks.

teeedubb, 2016-03-13:
Awesome, thanks very much. Scp keeps getting better and better 😀

tomahawk51, 2016-03-14:
I came here to suggest this improvement as well.  I have a dedicated PC that at this point does not require a mouse/keyboard from power-up EXCEPT for this pop-up.  I'd love to know when updates are there (though an option to suppress updates could make sense too), and happy to dig out the keyboard when there are some.

The related issue, I believe was already reported, is that we get update notifications when there is NO UPDATE.  Of course, this also could be improved (show only when there is an update).

I really thank you for this tool - it's brought an amazing amount of fun into my family room.

nefarius, 2016-03-14:
@tomahawk51 yeah, it's annoying to me too, sadly I pushed the last release without realizing this stupid dialog, next release will remain silent, I promise!

closed 2016-04-10 by nefarius