nefarius/ScpToolkit #236: Deadzone for normal button presses

SlowSeer, 2016-03-13:
When looking at the 360 button layout under Game Controllers (Devices and Printers, Game Controller Settings), X and R1 are being pressed frequently on their own, but when I check Input Tester in SCP Monitor the buttons look like they are fine, with no visible pressure on the display slider.

It's possible that my controller could be registering minuscule amounts of false pressure on the buttons that aren't visible on Input Tester since it has a sliding display, where the 360 layout is on/off.

It would be awesome to have a pressure sensitivity dead-zone for normal buttons too.

paulofoca, 2016-07-23:
My DS3 is also acting like this... I tried the old crappy "MotionInJoy" drivers with Better_DS3, and in it I was able to set the "analog to digital threshold" (a.k.a deadzone for buttons) and it is now working perfectly fine! Except that I'm stuck on this horrible driver instead of the one on ScpToolkit now.

AdrienCardoso, 2017-02-08:
Same here. Was about to disassemble my controller thinking that it was faulty.

closed 2018-01-26 by nefarius