nefarius/ScpToolkit #237: Controller wont charge :(

PoisonStardust, 2016-03-14:
Well, i have been using this app for several years now but just recently i installed the lastest release of this tool to give it a try, everything was going awesome until i had to charge my controller, it wont charge, i tried rolling back to the old but it wont help, it works just fine i can pair it and use it no probs, but when i plug it in it wont charge i can still use it as a wired controller as it hows like it is pluged in, but i repeat, it just wont charge, strange thing here is that i can charge another controller that i have using the same cable.

i have tried using different cables, reinstalling the tool, rolling back to old drivers. i'm using geniune controllers on windows 10 any help is much appreciated, thanks

charlesmilette, 2016-03-14:
Have you tried charging the faulty DS3 with a PS3?

You said it has been several years that you use this tool. Maybe your controller battery is too used to operate correctly or has reached is [maximum charge/discharge cycles](https://en.wikipedia.org/wiki/Battery_(electricity)#Battery_lifetime).

PoisonStardust, 2016-03-14:
i just tested it on another PC :/ and it charges just fine. i don´t know whats going on ._.

charlesmilette, 2016-03-14:
Could you provide debug info? Maybe @nefarius could look into it.

PoisonStardust, 2016-03-14:
[ScpToolkit_Log-Package_felix_1457924382.zip](https://github.com/nefarius/ScpToolkit/files/172275/ScpToolkit_Log-Package_felix_1457924382.zip)

Stoann, 2016-03-26:
I don't know why, but mine stopped charging after a while too. Someone offered the following solution which works for me. It's a little annoying, but if you don't have to charge too often it's fine:

> Go to your Device Manager and then to your Human Interface Devices section before plugging in your PS3 controller. Then plug in your PS3 controller and you will see "HID-compliant game controller" and "HID-compliant wireless radio controls" be updated to your Human Interface Devices section. Right click and disable "HID-compliant game controller" then unplug and replug in your PS3 controller and it should charge fine.
>

PoisonStardust, 2016-04-04:
i reinstalled windows and at first it looked like it worked ok,it charged completly, but after it depleted the battery, it wouldnt charge, i don´t know if its something i installed or what´s goin on, i tried it on a LAPTOP and it works fine. but in mine just won´t charge.

Jan200101, 2016-04-06:
Why is everyone saying its the drivers fault for not charging the controller anymore ? It would make sense if it maybe disabled the port but as you said it would charge at all so try getting another charger or try charging it on a console or something like that

PoisonStardust, 2016-04-06:
i´m not saying it´s the drivers fault, as i stated before, i don´t know what the main problem is. Also, i already said i tested on another PC working just fine, im not complaining here about anything, i just want some help. thanks for the reply tho.

bardozan, 2016-09-20:
It's an old thread but I had this problem and solved it. If anyone has also this problem, maybe this can help.
I had a problem activating DS3 on my computer. Somehow there was a error like "couldn't open device". In another thread one guy suggests to reset the DS3 by pushing the little button back side. I did it and it worked. My PC was able to recognise it, but this time it was not charging.
I had an idea connecting it to my PS3. Maybe it was related to resetting it or something, but it needed its mother device. I was able to charge it with PS3 and after that, my PC was also.
Long story short, connect your DS3 to a PS3 if you have one. See if it's charging and try it again on your PC. Hope that helps.

kharisma88, 2016-11-09:
You need to do disconnect with R1 L1 first while the usb is plugged and then press home button to connecting it again for charging function. However, it keeps disconnecting and stop charging if the controller is not being used because of iddle.

Flagonard, 2017-02-02:
I´m having this issue too, it appeared suddenly a while ago and the only solution was to use another pc or ps3 to charge, very annoying. I was using an old 1.2.2 ver when appeared, i tried to update to the latest 1.7 but the problem persist, i mainly use BT and the DS3, connecting the DS3 instead of the BT dongle shows the controller charging and active (ON, using power) but not how it should (OFF and charging). 

If Nefarius could check this would be awesome. The objective is to connect the DS3 without opening the server and keep the controller charging with all lights blinking as before (Like ps3), or same behavior with the server open.

Thanks.

Trevomader, 2017-03-16:
There seems to be a problem with the cables, on one cable i charge just fine on two others i can't charge at all giving me status "241" instead of "charging".

Both cables give this exact debug log in standalone server:

[image]

Yhatax, 2017-08-29:
I also got this problem a few days ago:
Charging -> 241
Scp version 1.2.0.160

After searching and getting some info, I remembered this happened some time ago and I managed to solve it pretty easy, but did not remember how. So for the record:

In my case it was as simple as pluging the controler cable to the USB port I originaly installed it, as simple as that. I cable managed and got the USB cable to another port, and when Battery run Low, problem appeared. Hope it helps to troubleshoot for future.

closed 2018-01-26 by nefarius