nefarius/ScpToolkit #239: BSOD right after plugging the controller in

alvarosg, 2016-03-15:
I would like to check if any of you has run into the same issue today.

I have been using a couple of fake controllers for (two months now) connected to the USB, up to yesterday night. Today when I came back from work I tried plugging one of them, and after a few second I got a BSOD (Particularly  DRIVER_IRQL_NOT_LESS_OR_EQUAL (ScpVBus.sys)). It kept happening, after a couple of restarts on different USB ports, so I figured it could be that the controller had broken, however, when I tried the other controller I got the exact same result.

I ran the wipe utility, and then nothing would happen after connecting the controllers, but as soon as I reinstalled the driver, the BSOD was back, without even plugging it out and it again. I tried uninstalling and downloading the latest version of WinSCP to do a fresh install, but it did not work.

SCPToolkit was definitely not updated in between, but I noticed that Win 10 actually did some updating, so maybe actually  Win10 installed something that made this stop working.

Have any of you seen this behavior?
[image]

nefarius, 2016-03-16:
A kernel crash dump would **really** help to debug this =/

charlesmilette, 2016-03-16:
Instructions on how to provide it would help.

shauleiz, 2016-03-16:
The procedure for finding the DMP files:
Navigate to C:\Windows\Minidump, use the latest file. If the minidump folder is not there or empty there may be a larger DMP file located at C:\WINDOWS called MEMORY.DMP which is OK.

charlesmilette, 2016-03-16:
Thanks for the 5 tips...

nefarius, 2016-03-16:
=D I think @shauleiz Browser is going wild or something...

shauleiz, 2016-03-16:
It is the bloody Chrome on iPad

alvarosg, 2016-03-16:
I have now attached the three last kernel crash dump in the first post

alvarosg, 2016-03-16:
Something else that I have noticed:
If I do a wipe clean, and then use the driver utility to only install the service, but I do not install the driver, the error still happens (it still detects the controller and goes blue screen).

I forgot to mention before that this (the BSOD) only happens after I press the PlayStation button (which I need to do for the controller to be assigned to a number).

charlesmilette, 2016-03-16:
I never had to press the PlayStation button on the controller for it to get assigned.

alvarosg, 2016-03-16:
I did not normally had to do it before the BSOD started happening, but now, if I do not do, the controller just does not work, and then when I press it, it gets assigned a number, and then very quickly goes to the BSOD

charlesmilette, 2016-03-16:
Probably that the Bluetooth is faulty. Since you press the PS button, it probably attempts to connect via Bluetooth.  

Is the Bluetooth driver installed on your computer?

alvarosg, 2016-03-16:
I do not even have a bluetooth adapter in the computer.

nefarius, 2016-03-16:
Since the BSOD happens because of `ScpVBus.sys` connecting the controller making it happen is only a side-effect since once the controller gets connected, the virtual X360 pad gets spawned. That's probably when things go wrong.

alvarosg, 2016-03-16:
So any idea for a quick fix, anything that I could try reinstalling or rolling back, that does not involve doing a windows restore to solve it for now, or do you think the next release will solve this?

Cheers :)

nefarius, 2016-03-16:
@alvarosg sure have one but first; what's your exact Windows version and architecture (64-Bit, 32-Bit)?

nefarius, 2016-03-16:
@alvarosg I just assume your system is 64-Bit so here we go:

1. Stop the ScpService.
1. Open up Windows Device manager, in the category `System` search for `Scp Virtual Bus Driver` and uninstall it.
1. Download this: [ScpVBus_NEW_BETA_x64.zip](https://github.com/nefarius/ScpToolkit/files/176962/ScpVBus_NEW_BETA_x64.zip)
1. Unblock it.
1. Extract it to some arbitrary location.
1. Open an elevated command prompt (Google).
1. Navigate into that folder (using `cd`, Google)
1. Execute `devcon install ScpVBus\ScpVBus.inf Root\ScpVBus`
1. Start the ScpService.
1. Cross fingers and test.

Thanks.

alvarosg, 2016-03-16:
That did it! Both controllers are now working as before :) You are a genius Nefarius :D

What was the fix? (Not that I have ever done anything with drivers, or know that much about how they work, but I am still a software engineer, more on the machine learning side hehe, but it is always interesting :) )

Thank you so much! 

PS: Do you accept donations? I would not mind buying you a beer :P

charlesmilette, 2016-03-16:
*Downloads and installs the beta driver*

alvarosg, 2016-03-16:
Of course, I mean, what was changed in the beta driver xD

charlesmilette, 2016-03-16:
A lot
https://github.com/nefarius/ScpToolkit/compare/v1.6.238.16010...master

nefarius, 2016-03-16:
@alvarosg great news to me! I basically updated the surrounding development kits to Microsofts latest and greatest ones for Windows 10, created a proper compiling environment by introducing a fresh Visual Studio Project and had some help from @shauleiz :) [I moved the bus driver to a separate repository btw.](https://github.com/nefarius/ScpVBus), there you can read on the commits and changes which happened since Scarlet.Crushs original code base.

alvarosg, 2016-03-16:
Nice! Thank you for the info and again, for your help :)

shauleiz, 2016-03-17:
@nefarius - using **devcon** may result in unpredictable Device Instance Path as shown in the attached picture:

[image]

I did not look into the code yet to see if you rely on the normal path (ROOT\SYSTEM\\**0002**) but either way, we need to prevent the user from installing a second Scp Virtual Bus.

nefarius, 2016-03-17:
@shauleiz yep, I also stumbled upon this. That's why I included the manual driver removal in my short installation instruction above. The installer of ScpToolkit takes this into account but I don't know if I can put some sort of protection into the driver itself to avoid this.

shauleiz, 2016-03-17:
@nefarius 
Right, I already found the function  _Initialize()_  in your new XOutput project. Also in _FindDevice()_
I think we can add protection against installation of a second Virtual Bus.
This can be done in the installer (Just call FindDevice(BUS-GUID)) or in the addDevice function (Bus_AddDevice) which is more complicated but prevents manual installation of a second bus.

nefarius, 2016-03-17:
@shauleiz this kind of protection is already included in the ScpToolkit Driver Installer. I created the `XOutput` project as a convenient user-mode library for interacting with the bus, abstracting away the `DeviceIoControl` calls. Still a work-in-progress though. As for doing this in the driver itself; is it save to call functions from `setupapi.lib` from a kernel-mode driver?

mirh, 2016-03-17:
Possibly related to #29 and #205

shauleiz, 2016-03-17:
@nefarius 

- Some setupapi-like functions can be called from a kernel driver. Take a look at [IoGetDeviceInterfaces](https://msdn.microsoft.com/en-us/library/windows/hardware/ff549186(v=vs.85).aspx) routine. I think it can be used for our needs.
- XOutput is a good starting point. I don't see that you use it (yet) but I am sure going to. Please take a look at my "Pool Requests" - you might want to integrate my changes/fixes into the code.

closed 2016-04-10 by nefarius

ski150169106, 2016-11-05:
The same problem happen to me even i tried the latest version Scp toolkit v1.7.277.16103 .Because i have no ScpVBus_NEW_BETA_x32.zip  so i can not follow your tutor because my ios is win7 32

ski150169106, 2016-11-05:
@nefarius

momogee, 2017-02-23:
Hi, I'm having a similar problem but doing the fix with ScpVBus_NEW_BETA_x64.zip doesn't work.  The devcon command results in this:

Device node created.  Install is complete when drivers are installed...
Updating drivers for Root\ScpVBus from D:\a\ScpVBus.inf.
devcon failed.

Scp Virtual Bus Driver does show up in my Device manager under system.
Then going through the process of installing the DS3 by usb (not bluetooth) results in BSOD with DRIVER_IRQL_NOT_LESS_OR_EQUAL (ScpVBus.sys) right after the service starts.

One question - you say step 4 - "Unblock it" and I'm not sure what that means.

Original problem was like this:
Scptoolkit was working fine by usb and wanted to try to get my bluetooth working.  The BSOD began when I tried Force Install using the older version of 1.2.0.160 with the 1.2.2.175 update.  Genius, using and older version.
Since then, I've tried uninstalling everything, clean install, Force Install using 1.6.238.16010, disabling the bluetooth device completely, nothing works.

I've attached my minidump.
[022117-8906-01.zip](https://github.com/nefarius/ScpToolkit/files/797411/022117-8906-01.zip)