nefarius/ScpToolkit #240: No way to see battery level?

f3flight, 2016-03-17:
In original SCPToolkit there was SCPMonitor which would show battery level ( I believe ) - I see on way to check battery level in the new version. Is it hidden somewhere or just not visible? Would be nice to have LEDs option in SCPSettings which would show battery level using LEDs (4 leds - full, 3 leds - 80% ... 0 leds = 20% or below). Would also be nice to just have an analog of SCPMonitor to see some connected gamepad data - macs, battery level, incoming event indicator.

What do you think?

nefarius, 2016-03-17:
ScpMonitor is still shipped even with my fork although a replacement for it is already in the works but nowhere near release state yet. You can set the LEDs to indicating the charging level in the Settings Manager.

f3flight, 2016-03-17:
@nefarius I tried that option in settings but I only get one LED flashing, same as if I choose so show port - will more leds flash when pad charges up, or how it will look like? I actually have a problem with them - they stopepd turning on via PS button. Since recently when I connect them to my laptop via USB they only blink once or twice and switch off. However if I connect them to my tablet (Windows too) they keep blinking and work while connected. I thought there's a problem with their batteries and wanted to see percentage but could not find it anywhere. I tried starting ScpMonitor exe from an earlier (original) version but it showed nothing - I installed the newer version without ScpMonitor.

Will try (re)installing your tool with ScpMonitor to see if it shows anything, also will check if I get more leds on on the pads if connected to my tablet for some time... Sucks that these DualShock 3 and Sixaxis do not charge from a usb charger...

rdrouin, 2016-07-15:
Wouldn't a simple notification do the job if your controller's battery went under let's say 15 %?

closed 2018-01-26 by nefarius