nefarius/ScpToolkit #243: Add support to any HID (Joystick, Wheel etc.)

shauleiz, 2016-03-21:
# Abstract:
Current implementation enables a hiding a Sony PS controller behind a virtual Xbox controller. I suggest expanding the support to any Human Interface Device (HID) that is attached to a USB bus.
# Discussion:
While ScpToolkit gives a fantastic solution to people possessing a SPS Controller,  it does not solve the problem for people possessing a standard HID and wishing to play an Xbox game. I believe that are far more such people than the ones possessing a SPS Controller.
The current ScpToolkit can be expanded to support any USB Human Interface Device (HID).
### Current State:
ScpToolkit client are **USB devices** that conform to a white-list of VID/PID that allow only certain devices.
The  USB constrain is due to the usage of library [HidSharp](http://docs.zer7.com/hidsharp/), which is a given limitation.
The white-list approach implemented in ScpToolkit in order to allow only SPS controllers.
### Enhancement:
In addition to the White-list, add an option to select an HID that is not on the White-list and is not a physical Xbox controller.
The change has to be implemented in the User-Interface and in ScpToolkit\ScpControl\Usb. Adding a generic device will has to introduce various mapping schemes following MS [document](https://msdn.microsoft.com/en-us/library/windows/desktop/hh405052(v=vs.85).aspx).
I suggest that by default, devices will be considered as [Gamepad](https://msdn.microsoft.com/en-us/library/windows/desktop/hh405052(v=vs.85).aspx#Gamepad) with an option for the user to select an alternate remapping scheme such as Arcade Stick/Arcade Pad, Wheel or Flight Stick.

charlesmilette, 2016-03-21:
Wow. This issue is well documented.

I'll add one thing. Is what you want is to convert [DirectInput](https://fr.wikipedia.org/wiki/DirectInput) (Generic Controllers) entries to [XInput](https://fr.wikipedia.org/wiki/Xinput) (Xbox 360 Controllers) ones?  

In that case, [xbox360ce](https://github.com/x360ce/x360ce) is what you are searching for.

charlesmilette, 2016-03-21:
ScpToolkit and relatives were created because the integrated generic controller driver did not output any controls, wether DirectInput or Xinput. Therefore, the PS3 controller is not HID-compliant.  
It requires a special driver for directly receiving communications from the controller and translating them into, in this case, Xinput entries, that can be understood by the computer.  
Any HID-compliant device will directly output DirectInput entries without any additionnal drivers or configuration. If you want an HID controller to be recognized as a Xbox 360 controller, x360ce was made for that (hence the name: "**X**box __360__ **C**ontroller **E**mulator")

shauleiz, 2016-03-21:
X360ce is a witty **workaround** to this issue.
It has its advantages an disadvantages. I still think that adding this will attract a huge number of new users.

charlesmilette, 2016-03-21:
You can:  

- Write your own implementation of ScpServer taking advantages of Scp's Virtual Bus and libusbK driver dedicated to convert DirectInput to Xinput.  

- Create a pull request adding the requested feature and a key mapping tool (I'd personnaly prefer to have a tool to customize all the key instead than only presets, because DirectInput devices vary waaaaaaaaaaay more than Xinput does).  

- Search thorougly on the Internet. I'm pretty sure you're not the only one disliking x360ce and somebody already wrote a driver for it. That's also how Scp was born: someone disliked Motioninjoy and wrote this tool. Only thing is that the driver is probably buried deep within search results.  

- See if @nefarius is willing to implement this feature.

charlesmilette, 2016-03-21:
Seems you don't even need to exit GitHub to find something:
https://github.com/Stents-/XOutput

shauleiz, 2016-03-22:
@charlesmilette 
You have give a very good description of the situation.
I was not aware of XOutput and it seems to be more or less what I had in mind.
Playing with it I found out that:

- The UI is too complicated
- It crashes.

Also, I still think that you need a unified tool for as many input device - why not connect PS3, PS4 and a standard Joystick through the same application?

> DirectInput devices vary waaaaaaaaaaay more than Xinput does

Not exactly. For each _Usage_, There is a limited set of possibilities. For example, a joystick may have up to 8 (Named) axes, up to 128 (numbered) buttons and up to 4 POVs (Either continuous or 5-position).
The mapping for these controls are prescribed by MS.

> Write your own implementation of ScpServer taking advantages of Scp's Virtual Bus and libusbK driver dedicated to convert DirectInput to Xinput.

I did write a prototype (Not using libusbK but rather dinput). The point is that I suggest integrating it into SCP. 

> See if @nefarius is willing to implement this feature.

Well, this is what I'm doing...

charlesmilette, 2016-03-22:
> a joystick may have up to 8 (Named) axes, up to 128 (numbered) buttons and up to 4 POVs (Either continuous or 5-position).  

That's exactly what can make it vary way more. A Xinput device is obligatory a Xbox 360 controller. But if a DirectInput device isn't a normal joystick or wheel defined by Microsoft and send custom entries? Generally, with programs using DirectInput, you can change the key bindings for that reason, while many with Xinput don't allow that, because they are sure you are using a controller with a predefined layout

Pxtl, 2017-03-03:
XOutput leverages an old version of SCPDriver that does not appear to have complete device support.  I'm using a set of Logitech PS3/USB devices ( 046d-cad2-PS3/USB Cordless Gamepad ) and XOutput only supports their first 3 axes, and its developer appears to have abandoned it.  Whatever is intercepting the inputs can only handle the first 3 axes instead of all 16 axes on this device.

Squall-Leonhart, 2017-04-15:
xoutput doesn't support 'special case' axes such as sliders, it is no different to x360ce in that regards.

you can replace the scp driver installed by xoutput with the one from scptoolkit, but then the xoutput utility becomes the limit, so still no support for sliders or vibration.

there are modifications versioned 0.12 and 0.22 which do not work properly,  0.12 is an unnecessary modification of 0.11 for xbcd support, which ends up spawning two devices instead of the just the one per controller  (0.11 works fine with xbcd, so i dn't know what the point was) and 0.22 axes are somewhat broken.

closed 2018-01-26 by nefarius