nefarius/ScpToolkit #244: Clean Windows 10 install - ScpToolkit doesn't seem to regonize my DS3 controller

Pajsmak, 2016-03-26:
Hi!

I reinstalled windows today and formated all my harddrives.
I tried to reinstall ScarletCrush's drivers and it didn't work, I saw your post on the forums and decided to try it.

The thing is, when I start ScpDriverInstaller.exe, my controller doesn't show up in the list of devices ([Screenshot](http://i.imgur.com/5hzw5Sh.png)) even though it is regonized by my computer ([Screenshot](http://i.imgur.com/r1BbSDI.png))

I ran the ScpDebugInfoCollector.exe, here is the ZIP file it generated:
https://www.dropbox.com/s/u6hdfzc38kqgjw4/ScpToolkit_Log.zip?dl=0

I have also run the CleanWipe, still didn't work :/

Thanks in advance for any answer!

Also, thank you for taking over this software and updating it, really made it nicer to work with! :)

xub1990, 2016-03-26:
I am sorry if this does seem like a stupid question but have you tried scrolling down in the list of devices shown in Driver Installer?

Pajsmak, 2016-03-26:
Yes I have :)

closed 2016-03-28 by Pajsmak