nefarius/ScpToolkit #245: Can't play wired without using the USB dongle at the same time

Wartogh, 2016-03-27:
Hi! In old versions - when it used the ScpToolkit Monitor (now legacy) -, I used it in wireless mode with a USB dongle and when I needed to recharge the battery of the DualShock 3, I could replace the USB Dongle by the USB Cable of the controller in the same USB Port and it worked, that is, I could play in wired mode while charging it's battery without the need of the USB Dongle to be plugged, and I could do both in the same USB port! Now in the last versions I can't do it anymore, that means I must plug the USB Cable in another USB port without remove the USB Dongle if I want to play while charging...

Does it's happening only to me? If not, can it go back to work as before?

Jan200101, 2016-03-28:
> "In the old versions - when it used the ScpToolkit Monitor"

Dont complaint over something in a old version and get the new one

Jan200101, 2016-03-28:
It works aswell you just have to set up the right USB ports

Wartogh, 2016-03-29:
How so? Can you be more specific, please? Do you mean that if I put the USB Dongle in the right USB Port, when I need to charge the battery of the controller I will be able to take off the USB Dongle and put the USB Cable in the same USB Port the dongle was, and be able to play?

If so, that doesn't make any sence to me since, like said, in the old versions it used to work the way I'm trying to do now, and in the same USB port i'm using now

Jan200101, 2016-03-30:
I mean (I dont know if this is still in the new code) if you put the Dongle in the front usb port install it and then switch it will not work.

closed 2018-01-26 by nefarius