nefarius/ScpToolkit #246: can't use my controller wirelessly

MWeWe, 2016-03-27:
they works with Scarlet.Crush old version
[ScpToolkit_Log-Package_WeWe_1459072471.zip](https://github.com/nefarius/ScpToolkit/files/190850/ScpToolkit_Log-Package_WeWe_1459072471.zip)

LOSSLOSS, 2016-03-27:
Check this: https://github.com/nefarius/ScpToolkit/issues/241
We'll have to wait for a fix.

On which version did it work?

MWeWe, 2016-03-27:
V1.2.2.175

Jan200101, 2016-03-28:
There wont be a fix for this one problem.
Its a problem with the service not starting right most at the startup on most systems so set it to late start

charlesmilette, 2016-03-28:
Or start it manually when needed.  
Anyways. Related to #48

MWeWe, 2016-03-29:
thank you guys,
can anyone tell me what should i do step by step

charlesmilette, 2016-03-29:
In a elevated command prompt, type `sc config Ds3Service start= delayed-auto`

Jan200101, 2016-03-30:
Or if you dont know how to do that i made a little Batch compiled into a Exe that does that and even tries to troubleshoot when its failed.

https://onedrive.live.com/redir?resid=531AF8D5D12A8FC8!2727&authkey=!AIKadP-HA816bn8&ithint=file%2cexe

by the way i was too lazy to make a name if you cant tell

MWeWe, 2016-04-09:
still having the same problem here my new files
[image]
[ScpToolkit_Log-Package_WeWe_1460203395.zip](https://github.com/nefarius/ScpToolkit/files/211242/ScpToolkit_Log-Package_WeWe_1460203395.zip)

now after reverting back to ScpServer V1.2.2.175
service not starting but when i open ScpServer.exe it works

closed 2018-01-26 by nefarius