nefarius/ScpToolkit #247: Screen flashes gray/black when I get my DS3 connected

ferausquin, 2016-03-27:
After a windows update I had to reinstall SCP drivers now when i connect my controller i hear the pop sound and instantly my screen goes black then gray then black and so on to eternity :/

nefarius, 2016-03-27:
[image]

ferausquin, 2016-03-27:
I'm sorry english is not my first language :(

nefarius, 2016-03-27:
No worries, I understood what you wrote, I just don't get how an USB plug-in event could mess up your screen...

ferausquin, 2016-03-27:
1. windows update
2. controller no longer works
3. had to uninstall and reinstall (updated it in the process) SCP Drivers
4. when I connect my controller everything goes black then gray and so on :/

ferausquin, 2016-03-27:
I don't get it either. I have search for this in the forums and here but got no luck :/

nefarius, 2016-03-27:
I don't have the slightest idea what to advise you; that just sounds too weird @_@

ferausquin, 2016-03-27:
Alright, well I'll try to fix that somhow if I discover what's wrong I'll post it here.

PD: Thanks for replying so fast! :O

ferausquin, 2016-03-27:
And thanks for your amazing work :)

nefarius, 2016-03-27:
Glad it's of use to you :)

RED-404, 2016-03-27:
This was a problem for me in Windows 10 build 14291 on 2 systems. Windows
10 build 14295 is rolling out now and will probably fix the problem.

nefarius, 2016-03-27:
@RED-404 I see Microsofts great plan of continuously throwing out high-quality updates really goes along well :D

RED-404, 2016-03-27:
Can't really hold this against them its an insider build & fast ring. Not
to mention this was in the release notes for build 14291.
"

   - Your PC may freeze when plugging in an Xbox One or Xbox 360 controller
   and other gamepads in this build."

nefarius, 2016-03-27:
@RED-404 ah, good to know :+1:

ferausquin, 2016-03-27:
Yeah I was just updating everything. I'm on build 14295 now, everything works great! :grinning:

charlesmilette, 2016-03-27:
Here is the slow ring Insider still on 10586 :/

CoolOppo, 2016-04-05:
> Yeah I was just updating everything. I'm on build 14295 now, everything works great!

I was having all the same exact issues you were having with the last build, so I rolled it back, and everything worked perfectly. Now, I just got build 14295, and I'm having issues again. It doesn't instantly cause the screen to go back and forth between black and gray any more, but the system starts lagging very badly, and continues getting worse until I unplug the controller. It gets so bad that the mouse cursor skips around the screen when I move it. Nothing appears to be using more resources than it should in Task Manager, so I have no idea what could be causing it.

ferausquin, 2016-04-05:
> I was having all the same exact issues you were having with the last build, so I rolled it back, and everything worked perfectly. Now, I just got build 14295, and I'm having issues again. It doesn't [...]

Ok, so it may not have go so smooth for me either.
I did experience lag. Over any gaming session the lag kept getting worse and worse from smooth to unplayable until my screen flashed black and it ran smooth again for a minute, then began lagging again.
I check'd my processes as I play with the DS3 pluged and unpluged and saw a tiny difference in the CPU usage.

Anyway, what I did was switch to motionjoy :( It sucks, but I haven't found other way to play without problems.

I hope this gets fixed on the Microsoft end sometime soon....

CoolOppo, 2016-04-06:
Yeah, I had to roll back to build 14279 again. It's odd that Microsoft would only partially fix an issue and then push the next version. I have got to get out of the Fast Ring.

charlesmilette, 2016-04-07:
That is what the Slow Ring is for !  
Altough I sometimes had to resist the urge to switch to Fast because of the new features... like now with build 14316.

CoolOppo, 2016-04-07:
> Altough I sometimes had to resist the urge to switch to Fast because of the new features... like now with build 14316.

I'm installing 14316 right now. I'll see if everything is working properly and report back.

charlesmilette, 2016-04-08:
So, @CoolOppo ? Any news?

CoolOppo, 2016-04-08:
> So, @CoolOppo ? Any news?

Yeah, sorry I forgot to reply! Unfortunately, the issue is not fixed in 14316 either; it's the same as it was in 14295 (causes instability and extreme lag). I have rolled back to 14279 again, and everything works great, although I really wish I didn't have to, because the Ubuntu command-line thing was absolutely incredible.

WilliamFeely, 2016-04-09:
For me, if I connect my DS3 controller to my AMD laptop, the performance deteriorates as CoolOppo mentions.  On my desktop, Fallout 4 (a game it can normally play quite well) starts to drop in FPS quite severely, but can be temporarily fixed by killing the dwm.exe process (I run it in borderless windowed mode).  Maybe there could be a GPU resource leak in the status app or something?

charlesmilette, 2016-04-09:
I don't know. But it's defenitely a driver bug, since some fixed it by downgrading the Xbox 360 controller driver to 2.1.0.1349 (https://www.reddit.com/r/Windows10/comments/4dmnza/announcing_windows_10_insider_preview_build_14316/d1t4235 and https://www.reddit.com/r/Windows10/comments/4dmnza/announcing_windows_10_insider_preview_build_14316/d1shjde)

charlesmilette, 2016-04-09:
Maybe try the same thing

WilliamFeely, 2016-04-23:
I just installed the 14328 build after updating to the latest ScpToolkit driver, and on my desktop so far so good, no performance deteriorations on my Core i7 desktop.
Appears the bug is now fixed.

ferausquin, 2016-04-23:
Great! I'll give it a try. Thanks for the update!

WilliamFeely, 2016-04-23:
Update:  Been playing Fallout 4 for a half hour, looks like all is good, after 3 bad builds.

nefarius, 2016-04-24:
Thanks for the update!

closed 2018-01-26 by nefarius