nefarius/ScpToolkit #250: Not working at All

HelpMe1234, 2016-03-29:
I recently upgraded to Windows 10 and uninstalled Motionjoy and I cannot figure out why the SCPToolKit program isn't working for me. I am using a DS3 (PS3) controller connected via USB. It is recognized as a PS3 controller by my PC. Here's a screenshot of the program when I try to install it.
[image]
There is no XBOX 360 controller listed under devices and hardware.

charlesmilette, 2016-03-29:
You need to have the driver installed before seeing any virtual controller

HelpMe1234, 2016-03-29:
The XBOX360 driver? That already comes with Windows 10 I thought.

charlesmilette, 2016-03-29:
The driver for the PS3 Controller.  
Check PS3 Controller on the driver installer and press "Install"

HelpMe1234, 2016-03-29:
Like in the picture above...?

charlesmilette, 2016-03-29:
Yes.

HelpMe1234, 2016-03-29:
I did... and the log is from after I did that. Still no XBOX360 controller in devices.

charlesmilette, 2016-03-29:
Have you checked in the control panel?  
[image]

HelpMe1234, 2016-03-29:
Yup Control Panel > Devices and Printers only shows PS3 controller no xbox controller.

charlesmilette, 2016-03-29:
Try unplugging and repluging your controller to see.  

If it won't work then try this:
- Unplug the PS3 Controller
- Delete the libusbK driver and remove it from the DriverStore
- Then plug your PS3 Controller and reinstall the driver

HelpMe1234, 2016-03-29:
So delete C:\Program Files\Nefarius Software Solutions\ScpToolkit\libusbK\amd64\libusbK.dll ?

charlesmilette, 2016-03-29:
No. Not a good idea.
Use [DriverStore Explorer](https://driverstoreexplorer.codeplex.com/). Find "libusbK" in the list, check it and press "Delete package"  

But before doing that, you'll need to delete the device
With the controller unplugged, go in the Devices Manager:
[image]
Then show hidden devices:
[image]
A category named "libusbK devices" should appear. Expand it, and "PLAYSTATION (R) 3 CONTROLLER" should appear (or something equivalent). Right-click on it and press "Uninstall". Accept the prompt and __leave "Delete driver package" unchecked__. Then proceed with DriverStore Explorer

HelpMe1234, 2016-03-29:
I have 3 different libusbK listed in Driverstore explorer, delete them all?

charlesmilette, 2016-03-29:
Can I have a screenshot?

HelpMe1234, 2016-03-29:
[image]

charlesmilette, 2016-03-29:
Yes. Delete them all. Also think to remove the MotionInJoy driver (at the bottom). Can cause some conflicts

HelpMe1234, 2016-03-29:
Ah I tried to delete everything from MotionJoy but it looked like I missed that. And now that I deleted all the libusbK drivers it no longer says "already using libusbk, installation aborted." in the log. But it still doesn't show any XBOX controllers. Also the log looks different. 
[image]

HelpMe1234, 2016-03-29:
Just uninstalled and installed again and this is what I got 
[image]

charlesmilette, 2016-03-29:
Now try to unplug and replug the controller.  

If it won't work, try to uninstall "PLAYSTATION (R) 3 Controller" from the HID devices in the Devices Manager (with the controller unplugged and hidden devices shown). This is the stock driver, that ... don't works. When you replug your controller, should recognize libusbK in the DriverStore and install it instead of the stock one.  

Will be offline for ~15 minutes. Don't expect an answer in this time lapse.

charlesmilette, 2016-03-29:
The controller is under Human Interface Devices.
Try reinstalling only the driver with "Force Driver Installation" checked

HelpMe1234, 2016-03-29:
Still nothing... 
[image]

charlesmilette, 2016-03-29:
I'm out of idea. What do you suggest, @nefarius ?

nefarius, 2016-03-29:
*Composing ideas*

HelpMe1234, 2016-03-29:
So in the log it said "Reboot Required". So, I restarted my computer and lo and behold... 
[image]

Thanks so much for the help charlesmilette! Very much appreciated.

nefarius, 2016-03-29:
*Aborting composing ideas* =D

charlesmilette, 2016-03-29:
Don't forget to close the issue.

closed 2016-03-29 by HelpMe1234