nefarius/ScpToolkit #251: SCP stopped working windows 10.

Speccu, 2016-03-30:
My DS3 was stuck on controller port 2 and I uninstalled two instances of xbox 360 controller from my devices in an attempt to set it to controller one, now scp doesn't work.  I've tried the cleanup utility, uninstalling and reinstalling and nothing seems to work.

Here is a screenshot of the scp monitor, my ds3 is on port 2 and doesn't show up when I try to configure emulators or any other program.  Port 1 is an arcade stick that was working also but now is not.

http://puu.sh/nZz1V/459a1f8e9b.png

[ScpToolkit_Log-Package_Zack_1459353761.zip](https://github.com/nefarius/ScpToolkit/files/195934/ScpToolkit_Log-Package_Zack_1459353761.zip)

Speccu, 2016-03-30:
Never mind, I fixed the issue by reinstalling to a folder outside of program files and a few windows resets.

Windows 10 sure is great having like 3 device managers

mirh, 2016-04-02:
So.. can it be considered fixed?

closed 2016-04-02 by nefarius