nefarius/ScpToolkit #253: Controller doesn't show in installer.

Devilicius, 2016-03-31:
On the little drop down to select the controllers to install; the controllers are non existent. Please help.
[image]

charlesmilette, 2016-03-31:
Have you tried to scroll?

Devilicius, 2016-04-01:
Here's the debug report.
[ScpToolkit_Log-Package_Devilicius_1459486145.zip](https://github.com/nefarius/ScpToolkit/files/199009/ScpToolkit_Log-Package_Devilicius_1459486145.zip)

charlesmilette, 2016-04-01:
The PID and VID given by your screenshot says you installed the Bluetooth driver on a Xbox 360 Controller.

charlesmilette, 2016-04-01:
Do the controller works by USB, at least

nefarius, 2016-04-01:
[image]

You better start all over...

Devilicius, 2016-04-02:
yea, had to restore and use previous working version of the installer. everything works now.

nefarius, 2016-04-02:
Soooo, issue can be closed then?

Devilicius, 2016-04-02:
yes

nefarius, 2016-04-02:
Then do it please to keep things tidy here... I did it for you.

closed 2016-04-02 by nefarius