nefarius/ScpToolkit #257: Wireless Not Working?

MuffinPirateX, 2016-04-03:
I selected all my controllers and my Bluetooth dongle at install, installed the drivers, but both my PS3 and PS4 controllers won't connect wirelessly and will still try to connect to the consoles.  Also this site is being dumb and won't let me send my zip file stating that it has to be a zip file... because logic.

gergesh, 2016-04-06:
Joining on this issue here. Installed everything, have a genuine DS3 and an ASUS BT400, and installed the drivers for both. When I start up the server it says the dongle supports both DS3 and DS4, and that a genuine DS3 was detected.  Wired works just dandy, but as soon as I plug out the controller the status in ScpMonitor changes to 'Reserved', while The LEDs on the controller are still working. Pressing buttons doesn't work.

Jan200101, 2016-04-06:
Its a problem with the service most of the time try running this tool made by me 
[Download](http://bit.ly/1N8ZtSC)

and in case you wanna be save I put the source code up on pastebin and put it in a antivirus search

[http://pastebin.com/raw/zR2CbsBM](Sourcecode)

charlesmilette, 2016-04-06:
Maybe also #51

Jan200101, 2016-04-06:
@charlesmilette my tools tries to fix it because its a simple error if the service not starting like it should

charlesmilette, 2016-04-06:
#51 is something completely separated. The services detects the Bluetooth card, but can't handle it correctly, so any wireless communication is totally not working. Happens on the latest stable.

gergesh, 2016-04-06:
@charlesmilette does it happen on the stable before that aswell? Can I just install that instead to fix the issue?

Jan200101, 2016-04-06:
@charlesmilette 

> It's definitely an issue with the software as the the Bluetooth card is still picked up by the computer but the software no longer recognizes it appropriately. 

In the driver installer you have to pick what Bluetooth card / dongle you want to use so its either a faulty bluetooth card or someone who doesnt know how to work with computers

charlesmilette, 2016-04-06:
@Jan200101 
Steps to reproduce #51:
1. Install the Bluetooth driver
2. Test the Bluetooth -- works
3. Restart
4. Test the Bluetooth --not working  

So,
> It's definitely an issue with the software as the the Bluetooth card is still picked up by the computer but the software no longer recognizes it appropriately.   

@gergesh I don't know. This issue seems more related to Windows 10 not doing stuff the same as 8 or 7.

gergesh, 2016-04-06:
@charlesmilette not error #51 for me. It doesn't work even straight after the installation of the drivers. Guess it's a W10 thing. Probably not the best place to ask, but are there any good alternatives to this software? I have four controllers waiting to be paired.

hermitmaster, 2016-04-07:
I'm having the same problem now... Bluetooth won't work at all. I get a code 10 on whatever Bluetooth device (dongle or built-in card) after installing drivers. I had a hell of a time getting the normal driver reinstalled and functioning afterwards. For the record, I originally installed in November and it worked fine until late February. One of the updates broke my setup.

gergesh, 2016-04-07:
Well, I'm (as usual) dumb. My PS3 was apparently on in the living room, and it was interfering with the connection between the PC and the controller. As soon as I shut it down the controller managed to connect to the PC.

Jan200101, 2016-04-21:
@charlesmilette there we go I am back from smashing my head against the table now back to you
It is not connecting to the controller because the service (the thing that the dongle needs to connect to the controller) is not starting like it should

charlesmilette, 2016-04-21:
#48 
Simply start it from the service manager, or type `sc start Ds3Service` in a elevated command prompt. You could also make a shortcut to `sc start Ds3Service` and check "Run as administrator"
[image]
PS: Best drawing skills ever

Jan200101, 2016-04-21:
@charlesmilette good that you now know that it wasnt the guys bluetooth card

mirh, 2016-07-05:
Try to put SCP service on "Late" start.

closed 2018-01-26 by nefarius