nefarius/ScpToolkit #258: Old ScpServer works but not the newer one

Lanqu, 2016-04-03:
I've been using old ScpServer with DS3 and everything works ok. When I use new ScpToolkit 1.6.238.16010 he at first few times just hangs. Now he clearly sees my DS3 as PLAYSTATION(R) Controller but even after installations done and service up and running I didnt see any XBOX controllers appeared. It is not recognized by Killer Instinct. And no tools like new ScpServer or ScpMonitor see it and no chance to make it work.
Though old ScpServer seems still being able to connect it as XBOX 360 controller. I thought that he shouldnt be able do that if Ive messing with new Scp toolkit with new drivers and so on and so  force. So maybe it just didnt delete all the stuff before new install? What can I do to setup my DS3 and DS4 pads working? With PS4 pad things going the same way =(

dafluke, 2016-04-03:
which version works for you? I have the same problem with the latest one.

Lanqu, 2016-04-03:
ScpServer (v1.2.2.175) works for DS3 as usually it does. Even clicking uninstall drivers cant make it stop working

dafluke, 2016-04-03:
yeah the old one does not work for me. i'm using windows 10 on a new dell xps laptop. I think I'm gonna give in and buy a xbone controller. what a waste of time. :(

Lanqu, 2016-04-03:
for me it works ok.. also new PC. I think you need to reinstall drivers or so...

closed 2018-01-26 by nefarius