nefarius/ScpToolkit #260: PC Remote Play compatibility

julianxhokaxhiu, 2016-04-06:
Now that this layer is finally available directly from Sony, do you think your driver will conflict with this package? Should I uninstall it in order to get the Remote Play working, or can I just install it straightforward?

Thanks in advance for any reply :)

Jan200101, 2016-04-06:
Remoteplay is like a game it sees your controller and uses it there will be no problems if the guys at sony arent dicks but I dont think they are since the presentation with the PCMR in it

charlesmilette, 2016-04-06:
Probably won't work if RemotePlay takes only PS4 controller, since the libusbK driver does not expose the control outputs. But if it supports Xinput, it will work out of the box.

nefarius, 2016-04-06:
I don't have the slightest idea how PC Remote Play will work but if it expects the DS4 to be connected via standard hidusb.sys, then no, they won't happily co-exist I'm afraid.

Jan200101, 2016-04-06:
It worked back on the PS3 it just had no use because it was just a feature forgotten

julianxhokaxhiu, 2016-04-07:
Thank you for the replies guys! I didn't though I would get so many honestly ahah

Unfortunately I didn't had the chance to try it yesterday, but I'll keep you posted maybe today evening/tomorrow maximum about it.

Anyway I was asking this question honestly because I'm using this driver for emulation purposes and it's working awesome, and I don't really like to switch between the two continuously...I hope that maybe one day @nefarius you'll find a way to workaround it in case it's conflicting. In case it's not better this way and this issue could be closed straightforward :D Thanks in advance to anyone!

mirh, 2016-04-08:
Considering it doesn't even support [W7](http://www.kotaku.com.au/2016/04/ps4s-remote-play-doesnt-support-windows-7/), I'm guessing they decided to go full MS-Vision with APIs and all.

secondfry, 2016-04-19:
It doesn't work with PS4 Remote Play as it's using different driver than expected by Remote Play app.

mirh, 2016-04-19:
Perhaps it's just a simple name check. 
I'd try to change that with x360ce.

Jan200101, 2016-04-21:
by the way @mirh it can work on every OS [_*Cough*_](http://www.tech-recipes.com/rx/7225/playstation-3-use-remote-play-on-any-non-sony-vaio-windows-7-pc-ps3-firmware-3-30/)

mirh, 2016-04-21:
I didn't know there was an official remote play program for PS3 too, thanks. 
We were talking of ps4 one here though.

charlesmilette, 2016-04-21:
Remote Play for PS3 doesn't support controllers.

Jan200101, 2016-04-21:
@mirh sadly the PS4 one only supports

- Windows 8.1
-Windows 10 or later
-OS X 10.10
-OS X 10.11

@charlesmilette 
It has no use anyway

Mahir007, 2016-04-24:
Remote play doesn't install or use any custom drivers as far as I know, at least it didn't install one, it simply said plug in DS4, I did and it worked just like that, but if you install SCP driver the DS4 doesn't work in remote play, even if you stop the service.

charlesmilette, 2016-04-24:
The thing that may make it work would be using the virtual bus to recreate a normal PS4 controller, and then install the regular Windows driver on it.

mirh, 2016-04-24:
I don't think you need anything that fancy. 
Though, I wonder if you couldn't have the DirectInput device (because that's what vanilla DS4 uses) expose the real thing. 
Ideally it shouldn't compromise XInput and 360 emulation.

nefarius, 2016-08-28:
**Update:** when [ViGEm](../../ViGEm) is finished this will be possible!

julianxhokaxhiu, 2017-07-02:
What's the current status? :D

julianxhokaxhiu, 2017-12-18:
@nefarius any update on this?

nefarius, 2017-12-24:
@julianxhokaxhiu It's already working perfectly with ViGEm & VDX.

julianxhokaxhiu, 2017-12-24:
Thank you very much :)

closed 2017-12-24 by julianxhokaxhiu