nefarius/ScpToolkit #262: Issues using controller wirelessly?

squack, 2016-04-08:
I have a genuine DS4 and it works fine (to an extent) when wired. The second i disconnect it to try and use it wirelessly SCP monitor shows it as "reserved" and the backlight turns into a bright grey color. At that point it stops working completely until I reconnect the controller. I also get a notification saying that SCP found a wireless controller but doesn't do anything about it?

Jan200101, 2016-04-21:
Its a problem many users have and its easy to fix (if you now a how to get into the deep stuff) but I kept you from that trouble and made this pile of code yay.
hope it fixes your problem (which it did in most other cases)

[My Fix](https://onedrive.live.com/redir?resid=531AF8D5D12A8FC8!2755&authkey=!AEphx3qq60WTNHs&ithint=file%2czip)

remivamp, 2016-04-26:
@Jan200101 All that is in that .zip file is a README.txt

konsumer, 2016-04-26:
Yeh, love to have a look at your solution.

Jan200101, 2016-05-13:
Just  noticed my mistake sorry for not responding I forgot my password I am working on a system that tries to check if it running at startup so I am going to make a new one and compile it

closed 2018-01-26 by nefarius