nefarius/ScpToolkit #263: Support for PS Move Controller

Nerdy314159265, 2016-04-09:
I just wanted to suggest the addition of PS Move Controller support at some point. Even if the 3D sensing can't be used, at least to begin with, it would be cool to have support for at least the other controls.

Heratiki, 2016-04-16:
The Move Navigation Controller as well.  It's practically useless in Windows 10 at this point.

nefarius, 2016-04-18:
I don't own those peripherals so can't make any promises unless someone else submits code or hardware.

Nerdy314159265, 2016-04-19:
@nefarius Do you have a P.O. Box or something for that sort of thing? I'd consider sending you mine after I find someone who can confirm it works. (I got it for $3 at a Goodwill) It charges fine and seems to search for a connection but we'll have to see what happens when I test it and how much shipping would cost.

Heratiki, 2016-04-19:
I'll look into it @nefarious and see if there is any code I could submit. Better to learn and assist than just assume someone else will do it. The Move Navigation Controller is recognized after you install the drivers for a standard DS3 but does not show up during the driver installation. 

The Move Controller itself shows nothing unfortunately.

Nerdy314159265, 2016-04-21:
I'd try and help but I don't know anything about driver coding. If you knew a handy guide, I'd get started and try to assist as well.

charlesmilette, 2016-04-21:
Same for me... along with C++ :smile:

zelmon64, 2016-05-22:
If you're only interested in the buttons, the ps move can be paired using the native Windows bt stack described in [this guide](https://github.com/HipsterSloth/psmove-unity5/wiki/PSMove-Setup-(Windows)). Windows will then recognise all the buttons.

If you also want IMU and positioning (requires a ps eye) information (plus trigger pressure) then there is a [FreePIE plugin](http://www.mtbs3d.com/phpbb/viewtopic.php?f=139&t=18265) under development (which works quite nicely imo =]). It also fully supports the Sharp Shooter and Racing Wheel attachments.

Heratiki, 2016-05-22:
> If you're only interested in the buttons, the ps move can be paired using the native Windows bt stack described in this guide. Windows will then recognise all the buttons.

Oh man!!! This like Christmas Comes early! Thank you!

Nerdy314159265, 2016-05-22:
@zelmon64 That first part is especially interesting. However, I think the downside of ScpToolkit is that it replaces the default Bluetooth stack for your adapter (if I remember correctly). So that means you can't use it and the PS Move at the same time.

zelmon64, 2016-05-22:
@Nerdy314159265  You're right about how it steals the adapter but if you have two it's fine. I have found that the cheap unbranded bluetooth 4.0 usb adapters work fine.

Heratiki, 2016-05-22:
I'll need to buy one on Amazon then because my standard BT 2.1 just isn't cutting it. Would you suggest I get two different Bluetooth manufacturers so that Scp doesn't overtake both accidentally?

zelmon64, 2016-05-23:
@Heratiki I just bought two of the cheapest I could find. They don't even have to be in different hubs (which I hadn't realised until now). I just tried it with one in Port_#0001.Hub_#0008 and the other in Port_#0002.Hub_#0008 (side by side in [this front panel sata hub](http://www.raidsonic.de/products/internal_cases/mm_cardreader/IB-864B/img/IB-864-B_info_big.png)) and it worked fine. Even randomly unplugging and plugging them back in didn't cause any problems. 

Another thing that surprised me was that I could replace the SCPtool one with a different unit and it still had the same mac address so I didn't have to repair the controller.

Heratiki, 2016-05-23:
@zelmon64 great I'll grab a couple and see how it goes!

mirh, 2016-08-28:
Ps move already has its working driver, together with a cool API
https://thp.io/2010/psmove/

Zer01neDev, 2017-04-16:
No it doesn't work, gyroscope data etc... is not send to windows.

zelmon64, 2017-04-16:
@Zer01neDev what are you trying to do? For the IMU data you either have to use PSMoveAPI like mirh said or use the newer [PSMoveService](https://github.com/cboulay/PSMoveService) (https://groups.google.com/forum/#!forum/psmoveservice).

Zer01neDev, 2017-04-16:
People doesn't want to make an app but use the PSMove as a directinput device, like every thing you connect on the PC.

mirh, 2017-04-16:
Well, surprise surprise, that ain't like everything you connect to a PC. 
You know, devices themselves know only a part of the equation at once. *Something* is needed to mix this information and expose it in a meaningful way to the OS.

Willguigo, 2018-01-06:
Hello @nefarius.
I'm testing Scptoolkit, is there any setup to use the DS3 motion sensors?

closed 2018-01-26 by nefarius